nasa/fprime #156: RPI: Not receving strings from RD_SendString (UART)

m-aleem, 2020-03-04:
I am following the steps from the RPI example README, using a Raspberry Pi 4 and Ubuntu 18.04 (on a VirtualBox VM). 

Per the board preparation, I ensured that Serial is enabled, but use of the UART for the Linux console

timcanham, 2020-03-04:
The README has some wire loopbacks you have to add. Did you do those?

https://github.com/nasa/fprime/blob/master/RPI/README.md

If you don't have the loopbacks, the strings won't echo.

m-aleem, 2020-03-05:
Thank you. I have not added those - I will give that a try and close the issue if it works.
 
Why does this prevent an Event message, just for the `RD_SendString` command? The` RD_SendSpi` command when sent still shows an event message (though it does not echo the string, just *****) - while the `RD_SendString` has no event message at all. Is this just a matter of different implementation?

timcanham, 2020-03-05:
It looks like the code doesn't issue the event:

```c++
  void RpiDemoComponentImpl ::
    RD_SendString_cmdHandler(
        const FwOpcodeType opCode,
        const U32 cmdSeq,
        const Fw::CmdStringArg& text
    )
  {
      Fw::Buffer txt;
      txt.setsize(text.length());
      txt.setdata((U64)text.toChar());
      this->UartWrite_out(0,txt);
      this->m_uartWriteBytes += text.length();
      this->cmdResponse_out(opCode,cmdSeq,Fw::COMMAND_OK);
  }
```
It looks like you found a bug. We define the event `RD_UartMsgOut` in the XML, but then never call it.

You want to fix it and do your first PR?

m-aleem, 2020-03-05:
Yes, will do.

So this should trigger a `RD_UartMsgOut` (Sent UART message) event. 

And, it appears that the `RD_UartMsgIn` (Received UART message) event is not issued and `RD_LastMsg` channel does not changed, so I figure `UartRead_handler` method is not running because of the aforementioned missing loopbacks.

Thank you!

m-aleem, 2020-03-09:
Fixed in PR 157

closed 2020-03-09 by m-aleem