nasa/fprime #162: Requesting support for debug info on invalid XMLs

LightPunster, 2020-03-17:
The implementation generation and build commands throw errors when an XML is not valid according to a schema; however, they do not provide either the type of error or the line number where the error occurred (i.e the line of the XML file), which makes it very difficult to debug. I have seen the following error:

`XML file <file-name>.xml is not valid according to schema <path-to-schema>/<schema>.rng.`

After using the `fprime-util impl` and `fprime-util build` commands for the following fairly diverse bugs:
- The specified path to the schema does not exist
- Comment tags are empty (`<comment></comment>`)
- Other tag groups are empty (`<parameters></parameters>`)
- Tag groups are not closed correctly (`<parameters>....<parameters>`)
- Tags are spelled incorrectly (`<perametrs></perametrs>`)
- Mismatched passive/active/sync/async component/port combinations (i.e. including an async port on a passive component or including no async port on an active component)
- A currently unknown Topology XML bug which is motivating me to submit issue.

It would be extremely helpful for debugging if the functionality was implemented to identify both which kind of bug has occurred and what line it has occurred on.

LeStarch, 2020-03-18:
We are working on a fix for this, which will report the error that the schema checker reports, as well as the line/column number where possible.

SterlingPeet, 2020-04-01:
This affects my team as well.

LeStarch, 2020-04-02:
I think we have a fix in progress for this.  If the schema checker reports a better error (column/line number) we will report that as opposed to just "Failed to validate against schema".

This should catch most of the errors and report them better.

LeStarch, 2020-05-26:
Delivered.  Let me know if this is insufficient.

closed 2020-05-26 by LeStarch