nasa/fprime #170: fprime/docs/INSTALL.md : Hanging on installing F' Python packages

jackyyl, 2020-05-15:
When I first attempted to install the F' Python packages with the commands: 
pip install ./Fw/Python
pip install ./Gds
It would hang halfway through the installation process. This was done on the Zynq-Z1 FPGA board. My current solution was to simply wait for however long it needed to finish running the command, in my case it was approximately 4 hours. I have tested the same command on WSL and had no issues installing the F' Python packages however.

LeStarch, 2020-05-26:
This likely is caused by a slow internet connection or caused by older python code.  When setting up python (before running the commands above), you can run the following to fix the problem.  We need to add these commands to the `INSTALL.md` document.

```
pip install --upgrade wheel setuptools pip
```

LeStarch, 2020-07-29:
Did this help? Can I close?

LeStarch, 2020-08-03:
Reopen if this hasn't helped you.

closed 2020-08-03 by LeStarch