nasa/fprime #171: Math Component F' Tutorial

csmith608, 2020-05-21:
After we have connected the components in section 3 of this tutorial we are told to run:
`fprime-util build`
from 
`Ref/Top`
If run from that location this error is thrown:
make: *** No rule to make target `Ref_Top'.  Stop.
[ERROR] CMake erred with return code 2.

I believe we should actually be running that command from 
`Ref`

LeStarch, 2020-05-21:
Yes, this needs to be fixed!

LeStarch, 2020-05-26:
Already fixed in devel.  Needs to be pushed to master for display.

LeStarch, 2020-05-26:
Delivered!

closed 2020-05-26 by LeStarch

Hackerman342, 2020-06-05:
Hi,
I'm getting this same error. Just to confirm - we should not be running `fprime-util build` from `Ref/Top` and should only run it from `Ref`?
Thanks.

LeStarch, 2020-06-06:
Yes that is correct.  A correction to the tutorial will be delivered in the next release.

Hackerman342, 2020-06-08:
Awesome, thanks!