nasa/fprime #173: Math Component F' Tutorial Section 4 GDS Error

csmith608, 2020-05-21:
In section 4 when the ground system is running and the MS_DO_MATH command is sent, the  error below is produced.  Other commands are sent and information shows up correctly.  
[image]

JHDeerin, 2020-05-28:
I'm getting the same error, but am also getting it for any command that takes a typed-in argument (e.g. `MR_SET_FACTOR_1`,  `factor2_PRM_SET`, `cmdDisp.CMD_TEST_CMD_1`, etc.). Math commands that don't take an argument, as well as commands with drop-down menus for selecting arguments, etc., seem to work fine and are recorded in the event log.

kbiesiadecki141, 2020-06-11:
Any update on this? I'm coming across this exact error only for commands that take arguments (mathSender.MS_DO_MATH, mathReceiver.MR_SET_FACTOR1, etc.) like JHDeerin above

LeStarch, 2020-06-12:
Yes.  It seems to be an issue with float and double arguments.   A fix is in progress.

LeStarch, 2020-06-12:
Once number #182 is merged, this problem should be fixed.  Nothing y'all did wrong.

LeStarch, 2020-08-03:
Closing this issue.

closed 2020-08-03 by LeStarch

lewisjaggi, 2020-10-13:
Hello,
I did the tutorial yesterday and I encountered this problem.
Can you tell me what can i do to fix it ?