nasa/fprime #176: [Math Component Tutorial] Unit test check fail

kbiesiadecki141, 2020-05-26:
Hi,

I'm running through the Math Component tutorial right now and while i've been able to successfully build (fprime-util build-ut in MathSender directory), my fprime-util check is failing. I seem to be getting different output than is shown in the tutorial, so I'm not sure if it's something in my code or how I'm running it. Any help would be much appreciated! 

[image]

LeStarch, 2020-05-26:
This looks like one your unit tests is failing.  You'll need to inspect the UT output to understand why. For your version of F´, the output is captured in a file.  You should be able to find the output here:
```
fprime/Ref/build-fprime-automatic-default-ut/Ref/MathSender/Testing/Temporary/LastTest.log
```

We are in the process of releasing F´ 1.4, which would display the UT output inline.   I can supply update instruction if that would help.

kbiesiadecki141, 2020-05-26:
Ah thank you!

LeStarch, 2020-05-26:
Before we close this issue, was there a problem in the tutorial that caused the error?  I want to make sure the tutorial isn't causing bugs.

kbiesiadecki141, 2020-05-26:
Nope, looks like it was a copy-paste error - accidentally had my completion status reply too soon. Thanks for a quick response!

LeStarch, 2020-05-27:
Ok, I will close.  I appreciate the feedback.  Welcome to F´!

closed 2020-05-27 by LeStarch