nasa/fprime #177: Installation Guide Clarifications (Virtual Env Troubleshooting, Werkzeug Import Error)

JHDeerin, 2020-05-26:
A couple clarifications/additions I thought might be helpful after running through the installation guide myself (I don't think any of these changes are urgent):

## Possible Clarifications

- Changing `cd <path/to/fprime/checkout>` to `cd <path to fprime checkout>`, to be clear that there isn't a folder called `fprime/checkout`
    - Again, this should already be clear to most Git users but IMO would be slightly clearer if changed

## Errors Encountered (Add to Troubleshooting?)

I ran the tutorial on a Ubuntu 18.04 VM (run via Vagrant/VirtualBox). The VM was run in headless mode (i.e. without a GUI, used via SSH).

- While installing F', I ran into an error when trying to create the virtual environment via `python3 -m venv ./fprime-venv` in a shared folder. The error read as follows:
    ```
    Error: [Errno 71] Protocol error: 'lib' -> '/vagrant_data/<...folder path...>/fprime-venv/lib64'
    ```
    I eventually solved this by following [this StackOverflow post](https://stackoverflow.com/questions/24640819/protocol-error-setting-up-virtualenvironment-through-vagrant-on-ubuntu) and creating the; I'm not sure if this error is Vagrant-specific or not, but creating the virtual environment folder in a non-shared folder fixed the error. 

    It might be worth noting to not try running this command in a shared folder if problems arise.
- While testing the installation by running `fprime-gds -g html -d <path/to/fprime/checkout>/Ref` after a successful build, I encountered the following error:

    ```
    Error: While importing "fprime_gds.flask.app", an ImportError was raised:
    
    Traceback (most recent call last):
    File "/home/vagrant/fPrimeTut/fprime-venv/lib/python3.6/site-packages/flask/cli.py", line 240, in locate_app
    __import__(module_name)
   File "/home/vagrant/fPrimeTut/fprime-venv/lib/python3.6/site-packages/fprime_gds/flask/app.py", line 12, in <module>
    import flask_uploads
    File "/home/vagrant/fPrimeTut/fprime-venv/lib/python3.6/site-packages/flask_uploads.py", line 26, in <module> from werkzeug import secure_filename, FileStorage
    ImportError: cannot import name 'secure_filename'

    [ERROR] Failed to run application: python3 -u -m flask run. Error: python3 stopped with code 2 sooner than 2 seconds.
    ```

    I'm still not sure what the solution to this is ([updating requirements.txt](https://stackoverflow.com/questions/60192172/importerror-cannot-import-name-filestorage-from-werkzeug) didn't fix it).

LeStarch, 2020-05-26:
I agree with your clarifications.  I have never seen these Python errors before....not sure what is going on with the second one.  Did you try running on a non-shared folder in the VM?

JHDeerin, 2020-05-26:
Yup, I ran it on the home folder (wasn't able to get the virtual environment working on the shared one); the 2nd error sounds like it has something to do with mismatched library versions (I'm not familiar enough with Flask/Werkzeug to know this for sure), but if it works on other computers I'm not sure why it'd fail on mine .

LeStarch, 2020-05-26:
This is an issue from where `flask-uploads` being installed from.

LeStarch, 2020-05-26:
I think I know what is going on....before you run your `pip install` commands, run:

```
pip install --upgrade wheel pip setuptools
```
Older versions of pip will not pull in flask-uploads from the right source.

LeStarch, 2020-05-26:
@JHDeerin can you test the above fix and report if it helped?

JHDeerin, 2020-05-26:
@LeStarch Just tried it and reinstalling the python packages/rebuilding per the guide, and got the same error ☹️. If it's significant, I ran everything while inside the virtual environment.

Here's the full output from that last step:
```
[INFO] Ensuring TCP Server is stable for at least 5 seconds
[INFO] Running Application: python3
[INFO] Log File: /home/vagrant/fPrimeTut/fprime/Ref/logs/2020_05_26-22_20_31/ThreadedTCP.log
[INFO] Ensuring comm[ip] Application is stable for at least 1 seconds
[INFO] Running Application: python3
[INFO] Ensuring Ref Application is stable for at least 1 seconds
[INFO] Running Application: /home/vagrant/fPrimeTut/fprime/Ref/bin/Linux/Ref
[INFO] Log File: /home/vagrant/fPrimeTut/fprime/Ref/logs/2020_05_26-22_20_31/Ref.log
[INFO] Ensuring HTML GUI is stable for at least 2 seconds
[INFO] Running Application: python3
 * Serving Flask app "fprime_gds.flask.app"
 * Environment: production
   WARNING: This is a development server. Do not use it in a production deployment.
   Use a production WSGI server instead.
 * Debug mode: off
Usage: python -m flask run [OPTIONS]

Error: While importing "fprime_gds.flask.app", an ImportError was raised:

Traceback (most recent call last):
  File "/home/vagrant/fPrimeTut/fprime-venv/lib/python3.6/site-packages/flask/cli.py", line 240, in locate_app
    __import__(module_name)
  File "/home/vagrant/fPrimeTut/fprime-venv/lib/python3.6/site-packages/fprime_gds/flask/app.py", line 12, in <module>
    import flask_uploads
  File "/home/vagrant/fPrimeTut/fprime-venv/lib/python3.6/site-packages/flask_uploads.py", line 26, in <module>
    from werkzeug import secure_filename, FileStorage
ImportError: cannot import name 'secure_filename'

[ERROR] Failed to run application: python3 -u -m flask run. Error: python3 stopped with code 2 sooner than 2 seconds. [INFO] Shutting down F prime due to error. Failed to run HTML GUI
```

LeStarch, 2020-05-26:
But did you run the following?

```
pip install --upgrade wheel pip setuptools
```

JHDeerin, 2020-05-26:
Yes, before re-installing the python packages

LeStarch, 2020-05-26:
Ok, try this....but you didn't create a *new* virtual environment?  If not:

```
pip uninstall file_uploads
pip install --upgrade flask_uploads @ git+https://github.com/maxcountryman/flask-uploads@f66d7dc93e684fa0a3a4350a38e41ae00483a796
```

LeStarch, 2020-05-26:
I see the thumbs up, did that help?

JHDeerin, 2020-05-26:
Yup, that fixed things! Thank you!

LeStarch, 2020-05-26:
So for those who find this ticket again, here is the cause:

1. flask-upload's latest supported version is hosted on github
2. We install it with a setup-tool "@ git+https..." location
3. Older version of PIP fail to read that and install from PyPI
4. Updating `pip setuptools wheel` are all required
5. The next release of `INSTALL.md` will include this upgrade step, and will remove this issue

LeStarch, 2020-05-26:
@JHDeerin good catch. Thanks for all of this!

aogrcs, 2020-09-08:
> ....but you didn't create a _new_ virtual environment? If not:
> 
> ```
> pip uninstall flask-uploads
> pip install --upgrade git+https://github.com/maxcountryman/flask-uploads@f66d7dc93e684fa0a3a4350a38e41ae00483a796
> ```

It did well

LeStarch, 2020-11-09:
Fixed.

closed 2020-11-09 by LeStarch