nasa/fprime #186: [Bug] Dictionary option in fprime-gds doesn't work

JHDeerin, 2020-06-18:
## Bug Description

Despite the help text saying the `--dictionary` option for `fprime-gds` "Overrides deploy if both are set", the `--deploy` option actually takes precedence right now - and since the deploy option has a default value, *this means the dictionary option is never actually used.*

## Steps to reproduce

1. Install F' per the installation instructions.
2. `cd` into the `Gds` folder.
3. Run the command `fprime-gds --dictionary ../Ref/Top/RefTopologyAppDictionary.xml`; it currently fails with the following error message:

```bash
[ERROR] Failed to pars arguments: F prime binary 'Gds' not found in <user's F' installation path>/fprime/Gds. Specify with '--app'
```

## Suspected Bug Location

The relevant bug appears to be in [lines 436-443 of cli.py](https://github.com/nasa/fprime/blob/de2e42157067ca8043d1ce764a000fa510ca8d48/Gds/src/fprime_gds/executables/cli.py#L436), as shown below:

```python
      # Find dictionary setting via "dictionary" argument or the "deploy" argument
      if args.dictionary is not None and not os.path.exists(args.dictionary):
            (...)
      elif args.deploy is not None:
            xml_dict = ParserBase.find_in(".*Dictionary.xml", args.deploy, True)
            (...)
```

Looking at this code carefully, we can see that the first `if` only triggers if the dictionary has a value *and* is an invalid path; the 2nd condition will trigger whenever deploy has a value, *including if the dictionary option is valid,* meaning the dictionary path will always be overwritten. As stated before, deploy currently defaults to the user's current working directory if not given; this means that, in practice, there is no possible path through this code where `dictionary` gets used.

## Suggested Solution

Add an additional check to the `elif` clause, as follows:

```python
(...)
elif args.deploy is not None and not args.dictionary:
    (...)
```

LeStarch, 2020-06-18:
@JHDeerin can you add this as a bug report template to the wiki?  This is a good format.

JHDeerin, 2020-06-18:
@LeStarch I would, but I can't see an edit button anywhere on the wiki page (it might need someone with write permissions)

LeStarch, 2020-06-18:
@JHDeerin will you go ahead and fix this for your CLI tool PR?

JHDeerin, 2020-06-22:
@LeStarch Sure; I'll go ahead and add the fix presented here.

closed 2020-07-14 by LeStarch