nasa/fprime #193: Enum not added to TopologyAppDictionary

sommercharles, 2020-07-17:
Hi,

I've specified an enumeration in xml (MMAMotorFaultEnumAi.xml) so I can use it with several variables of a serializable.
The binary is built and installed without errors. But when I launch gds, I run into the following error:

`fprime_gds.common.data_types.exceptions.GseControllerParsingException: Parsing error: Could not find type Ref::MMAMotorFault!`

Manually adding the definition of my enum in RefTopologyAppDictionary.xml does the trick.
But I'm guessing if the parsers don't add the enum automatically, I must have missed something, correct?

sommercharles, 2020-07-21:
Here is an attempt to better explain this issue using the Math tutorial and based on JHDeerin issue #186 template.

### Bug description:

Importing and using an xml enum type in a serializable successfully compiles, but does not create top dictionary enum entry, leading to GDS unable to find the referenced enum.

### Steps to reproduce bug:

Clone and install F' master branch per installation instructions
Follow Math tutorial up to the end
Add MathOpSerEnumAi.xml in MathTypes folder with the following code:

	<enum namespace="Ref" name="MathOpSer">
	    <item name="ADD"/>
	    <item name="SUB"/>
	    <item name="MULT"/>
	    <item name="DIVIDE"/>
	</enum>

Add MathOpSerEnumAi.xml in CMakeLists.txt file of MathTypes:

	set(SOURCE_FILES
	  "${CMAKE_CURRENT_LIST_DIR}/MathOpSerializableAi.xml"
	  "${CMAKE_CURRENT_LIST_DIR}/MathOpSerEnumAi.xml"
	)
	register_fprime_module()

Include enum and replace declaration of member "op" of MathOpSerializableAi.xml by MathOpSer:

	<serializable namespace="Ref" name="MathOp">
	    <import_enum_type>Ref/MathTypes/MathOpSerEnumAi.xml</import_enum_type>
	    <comment>
	    This value holds the values of a math operation
	    </comment>
	    <members>
	        <member name="val1" type="F32"/>
	        <member name="val2" type="F32"/>
	        <member name="op" type="Ref::MathOpSer"/>
	        <member name="result" type="F32"/>
	    </members>
	</serializable>

Add MathOpSer reference in Ref/MathReceiver/MathReceiverComponentImpl.cpp mathIn handler:

 ```
 void MathReceiverComponentImpl ::
    mathIn_handler(
        const NATIVE_INT_TYPE portNum,
        F32 val1,
        F32 val2,
        MathOperation operation
    )
  {
    // declare result serializable
    Ref::MathOp op;
    F32 res = 0.0;
    switch (operation) {
        case MATH_ADD:
            op.setop(MathOpSer::ADD);
            res = (val1 + val2)*this->m_factor1;
            break;
        case MATH_SUB:
            op.setop(MathOpSer::SUB);
            res = (val1 - val2)*this->m_factor1;
            break;
        case MATH_MULTIPLY:
            op.setop(MathOpSer::MULT);
            res = (val1 * val2)*this->m_factor1;
            break;
        case MATH_DIVIDE:
[... 98 lines cut ...]
                            for enum_file in enum_file_list:
                                enum_file = search_for_file("Enum", enum_file)
                                enum_model = XmlEnumParser.XmlEnumParser(enum_file)
                                enum_elem = etree.Element("enum")
                                enum_type = enum_model.get_namespace() + "::" + enum_model.get_name()
                                enum_elem.attrib["type"] = enum_type
                                members_elem = etree.Element("members")
                                for (member_name1, member_value1, member_comment1) in enum_model.get_items():
                                    enum_mem = etree.Element("item")
                                    enum_mem.attrib["name"] = member_name1
                                    # keep track of incrementing enum value
                                    if member_value1 != None:
                                        enum_value = int(member_value1)

                                    enum_mem.attrib["value"] = "%d"%enum_value
                                    enum_value = enum_value + 1
                                    if member_comment1 != None:
                                        enum_mem.attrib["description"] = member_comment1
                                    enum_elem.append(enum_mem)
                                    enum_list.append(enum_elem)

### Comment:

There's another issue linked to this one if I try to use an included enum straight into an element of a component (instead of using type="ENUM" etc.): the header of the enum is generated, but it is not included in the autocoded component header. 
Can we use enums like this?

LeStarch, 2020-07-21:
@sommercharles do you want to make a PR for the suggested fix?

sommercharles, 2020-07-21:
@LeStarch sure thing. I'll raise another issue for the header inclusion bug I mentioned once I have a fix for it.

sommercharles, 2020-07-23:
So regarding the other issue I mentioned, I browsed through the autocoder and it looks like I might be misusing enums in this case.

It doesn't seem that you can use import_enum in a component and use it as is.
A workaround I found is declaring a serializable with a single enum member and use that serializable several time through the component.

Too bad that we cannot re-use ENUM over several component members. Maybe someone can confirm?

Other than that, I think we can close this issue.

LeStarch, 2020-07-23:
This is an issue, and we'll fix it.

sommercharles, 2020-07-23:
Got it, thank you.

LeStarch, 2020-08-03:
Closing this issue.  Tracking bug separately.

closed 2020-08-03 by LeStarch