nasa/fprime #204: Default sizes of CMD_DISP table and SIMPLE_REG entries

sommercharles, 2020-08-08:
These two parameters are declared by default with these values:
```
#define CMD_DISPATCHER_DISPATCH_TABLE_SIZE 100
#define FW_OBJ_SIMPLE_REG_ENTRIES 500
```

I had to raise these parameters to respectively 200 and 600 (for now) because one of our components has 70+ commands and we also have a lot of driver instances.

Is there a maximum value for these parameters, or does it solely depend on the amount of memory the resulting FSW can access?
Would you recommend limiting the amount of commands and instances if possible?

LeStarch, 2020-08-10:
I don't know of a maximum value.  It will be limited by two things that I can think of:

1. Available RAM on the system
2. C++ array max index sizes

I don't specifically recommend slimming commands and instances. By I would recommend you think a bit about how someone will issue these commands to these instances as 200 commands is a lot to sort to.

sommercharles, 2020-08-10:
Sounds good, thanks. I'll have to check the max index size.
Yes, totally agreed on the number of commands, it's way too much to sort to. 
These commands implement all the set/get that a subsystem library offers, and there's a chance we only use a few. I implemented them in case we realize we need one of them during flight, for troubleshooting for instance. 
There's probably a better way to do this.

LeStarch, 2020-09-23:
Closing as it seems an answer was yielded.  Feel free to use our mailing list if you'd like to discuss more with a larger community: https://groups.google.com/g/fprime-community

closed 2020-09-23 by LeStarch