nasa/fprime #206: Passing -D Variables to CMake via fprime-util

SterlingPeet, 2020-08-11:
For projects where passing ``-D`` variables to the cmake call is necessary, the ``fprime-util`` tool needs a mechanism to send those flags through to cmake.  This is the recommended invocation (and the current output):

```
(build-osx_fprime-venv) ae-glrg-205034l:AssertDemonstrator speet3$ fprime-util generate Arduino -DARDUINO_SDK_PATH=/Users/speet3/src/Arduino/Arduino-1.8.12.app/Contents/java -DARDUINO_BOARD_NAME=ATmega128 -DARDUINO_FREQ=7372800
usage: fprime-util [-h]
                   {generate,purge,build,impl,impl-ut,build-ut,check,check-all,install,build-all,hash-to-file}
                   ...
fprime-util: error: unrecognized arguments: -DARDUINO_SDK_PATH=/Users/speet3/src/Arduino/Arduino-1.8.12.app/Contents/java -DARDUINO_BOARD_NAME=ATmega128 -DARDUINO_FREQ=7372800
```

To be clear, this is _not_ about attempting to pass compile defines directly from ``fprime-util`` through to the compile step, but specifically about making adjustments to the underlying ``cmake`` invocation.  Of course, that mechanism could be used to set compiler directives, but that should be up to the developer to write into the appropriate ``CMakeLists.txt`` files.

LeStarch, 2020-08-12:
Agreed.

LeStarch, 2020-09-23:
Fixed in progress.

LeStarch, 2020-11-09:
These work now.  Full functionality provided by 1.5.3.

closed 2020-11-09 by LeStarch