nasa/fprime #207: TopoFactory.py highest command ID check has no None check

AaronMcDaniel, 2020-09-01:
Autocoders/Python/src/fprime_ac/models/TopoFactory.py:222 has no None check in case highest_id was not assigned. This is different from the checks on lines 193 and 207. I had a deployment that failed because it had no events or telemetry and only had commands. 
I fixed it by adding a None check like on the other lines. Should this change be made or is it there for a reason?
I can make a PR if it helps explain what I'm questioning and suggesting.

LightPunster, 2020-09-13:
I had a similar issue with my deployment and can confirm that this suggestion works. I'd like to add that the correction should also be made on lines 238, 254, and 269 in order to support components with parameters but no events, telemetry, or commands.

AaronMcDaniel, 2020-09-14:
I added PR #209, which includes LightPunster's recommended fixes as well

LeStarch, 2020-11-09:
Merged in.

closed 2020-11-09 by LeStarch