nasa/fprime #212: Issue with sending floating point command arguments in GDS

abcouwer-jpl, 2020-10-06:
I'm branching off the latest master and adding my own Topology/Components. When trying to run a command with 8 F32 or F64 arguments ( I tried both ) I would get the following issue:

```
[2020-10-06 10:35:01,923] ERROR in app: Exception on /commands/myComponentImpl.MyProblematicCommand [PUT]
Traceback (most recent call last):
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/flask_restful/__init__.py", line 468, in wrapper
    resp = resource(*args, **kwargs)
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/flask/views.py", line 89, in view
    return self.dispatch_request(*args, **kwargs)
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/flask_restful/__init__.py", line 583, in dispatch_request
    resp = meth(*args, **kwargs)
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/fprime_gds/flask/commands.py", line 105, in put
    self.sender.send_command(command, arg_list)
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/fprime_gds/common/pipeline/standard.py", line 133, in send_command
    cmd_data = fprime_gds.common.data_types.cmd_data.CmdData(tuple(args), command_template)
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/fprime_gds/common/data_types/cmd_data.py", line 80, in __init__
    raise CommandArgumentsException(errors)
fprime_gds.common.data_types.cmd_data.CommandArgumentsException:  __init__() missing 1 required positional argument: 'typename' __init__() missing 1 required positional argument: 'typename' __init__() missing 1 required positional argument: 'typename' __init__() missing 1 required positional argument: 'typename' __init__() missing 1 required positional argument: 'typename' __init__() missing 1 required positional argument: 'typename' __init__() missing 1 required positional argument: 'typename' __init__() missing 1 required positional argument: 'typename'
```

Looking at the traceback, fprime-venv/lib/python3.6/site-packages/fprime_gds/common/data_types/cmd_data.py", line 80, in `__init__` is the following:

https://github.com/nasa/fprime/blob/master/Gds/src/fprime_gds/common/data_types/cmd_data.py#L80 

        errors = []
        for val, typ in zip(self.arg_vals, self.args):
            try:
                self.convert_arg_value(val, typ)
                errors.append("")
            except Exception as exc:
                errors.append(str(exc))
        # If any errors occur, then raise a aggregated error
        if [error for error in errors if error != ""]:
            raise CommandArgumentsException(errors)              <----- ERROR here
            
            
Looking at convert_arg_value, I saw:

https://github.com/nasa/fprime/blob/master/Gds/src/fprime_gds/common/data_types/cmd_data.py#L168 

    def convert_arg_value(self, arg_val, arg_type):
        if arg_val is None:
            raise CommandArgumentException('Argument value could not be converted to type object')
        if type(arg_type) == type(BoolType()):
            if arg_val == "False":
                av = False
            else:
                av = True
            arg_type.val = av
        elif type(arg_type) == type(EnumType()):
            arg_type.val = arg_val
        elif isinstance(type(arg_type), (F64Type, F32Type)):      <-------  SUSPICIOUS1
            arg_type.val = float(arg_val)
        elif isinstance(arg_type, (I64Type, U64Type, I32Type, U32Type, I16Type, U16Type, I8Type, U8Type)):
            arg_type.val = int(arg_val, 0)
        elif type(arg_type) == type(StringType()):
            arg_type.val = arg_val
        elif type(arg_type) == type(SerializableType()):          <-------  SUSPICIOUS2
            pass
        else:
            raise CommandArgumentException('Argument value could not be converted to type object')

I changed line SUSPICIOUS1 to 

        elif isinstance(arg_type, (F64Type, F32Type)): 

to match the 2 lines down for the integers. **Then I was able to successfully send my command.**

I think at line SUSPICIOUS2, calling `SerializableType()` with no arguments is an issue, because `__init__()` for `SerializableType` requires a `typename `parameter, and there is no default. I think that's what's getting reported when the if float case doesn't apply, and `SerializableType()` is called farther down.

https://github.com/nasa/fprime/blob/master/Fw/Python/src/fprime/common/models/serialize/serializable_type.py#L31 

So maybe there's two issues: 1. What's with `elif isinstance(type(arg_type), (F64Type, F32Type)):`, should `type()` be removed? and 2, should `SerializableType __init__()` have a default argument for `typename`?

Hope that's a decent writeup, Python is not my specialty.

abcouwer-jpl, 2020-10-06:
I restored `elif isinstance(type(arg_type), (F64Type, F32Type)):`, gave `SerializableType` 's `__init__()` argument `typename` a default of `""` and tried again. I got an error:

```
[2020-10-06 12:23:36,599] ERROR in app: Exception on /commands/motImpl.MotStartAllPosTraj [PUT]
Traceback (most recent call last):
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/flask/app.py", line 1950, in full_dispatch_request
    rv = self.dispatch_request()
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/flask/app.py", line 1936, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/flask_restful/__init__.py", line 468, in wrapper
    resp = resource(*args, **kwargs)
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/flask/views.py", line 89, in view
    return self.dispatch_request(*args, **kwargs)
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/flask_restful/__init__.py", line 583, in dispatch_request
    resp = meth(*args, **kwargs)
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/fprime_gds/flask/commands.py", line 105, in put
    self.sender.send_command(command, arg_list)
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/fprime_gds/common/pipeline/standard.py", line 133, in send_command
    cmd_data = fprime_gds.common.data_types.cmd_data.CmdData(tuple(args), command_template)
  File "/mnt/c/Users/abcouwer/fprime-venv/lib/python3.6/site-packages/fprime_gds/common/data_types/cmd_data.py", line 80, in __init__
    raise CommandArgumentsException(errors)
fprime_gds.common.data_types.cmd_data.CommandArgumentsException: Argument value could not be converted to type object Argument value could not be converted to type object Argument value could not be converted to type object Argument value could not be converted to type object Argument value could not be converted to type object Argument value could not be converted to type object Argument value could not be converted to type object Argument value could not be converted to type object
```

which matches what you'd expect if `convert_arg_value()` got all the way down to the else case.

LeStarch, 2020-10-06:
Is your test to use commands that gave F32 or F64 arguments?  In the upcoming release we corrected a lot of this imprecise type handling, so I'd like to see if this is repeatable there.

abcouwer-jpl, 2020-10-06:
> Is your test to use commands that gave F32 or F64 arguments?

I wasn't really doing a test per se, I was trying to implement a component and ran into this.  I tried two versions of the command, one with F32s, and one with F64s. Both ran into the same issue. When I changed the gds python files mentioned, both F32 and F64 command versions worked fine. 

> In the upcoming release we corrected a lot of this imprecise type handling, so I'd like to see if this is repeatable there.

Awesome. When is the release anticipated?

LeStarch, 2020-10-06:
Right!  I think in terms of tests, because I need to replicate the behavior to fix it.

We are targeting this month for the release.  A release announcement will be provided here when it is done: https://groups.google.com/g/fprime-community/

LeStarch, 2020-11-09:
Fixed in previous release.

closed 2020-11-09 by LeStarch