nasa/fprime #220: Write an Adapter and Framer Plugin Setup

LeStarch, 2020-10-22:
We should add an comm adapter and framer plugin setup for instantiating user's adaptations easily.

LeStarch, 2020-10-22:
Chat with GT for their approaches.

efriesema, 2020-10-25:
High LeStarch  I'm a Computer Science grad student working on a CubeSat project. Is somebody running this issue that they could use some help?  I know Python and C++ and I'd be willing to help make this happen. .My email is rononfilms01@gmail.com

LeStarch, 2020-10-26:
I don't have an assignee for this ticket yet. Another user recommended we contact them for advice on "what works" for python plugin stuff before starting.

efriesema, 2020-10-26:
Would you like me to follow up with that user and get a baseline idea of
the architecture and framework?  Do we have a basic abstract of what we
want the plugin to be able to do?
Ed Friesema
Check out my github page www.github.com/efriesema for my latest projects!

closed 2021-12-20 by LeStarch