nasa/fprime #229: Cannot pass arrays by reference

nathancheek, 2020-10-30:
I would like to pass an array by reference through a port, like such:

`<arg name="adcCounts" type="U16[8]" pass_by="reference"></arg>`

But the autocoder naively takes the `U16[8]` and turns this into `void invoke(U16[8] &adcCounts);` which fails to compile with this error:

`error: expected ',' or '...' before '&' token`

The proper format for such a function header is `void invoke(U16 (&adcCounts)[8]);`. It would be nice if the autocoder was extended to understand this special format used for arrays in function headers.

LeStarch, 2020-10-30:
Have you tried the new wrapper Array type?  

https://nasa.github.io/fprime/UsersGuide/dev/xml-specification.html  See "XML-Specified Array Type".

Signal Gen comes with several examples: https://github.com/nasa/fprime/tree/master/Ref/SignalGen

LeStarch, 2020-10-30:
Note: this suggestion is for passing in XML designed types that represent arrays.  Is passing in an array as a raw C++ type something you need?

LeStarch, 2021-12-20:
Closing this.  XML Arrays recommended.

closed 2021-12-20 by LeStarch