nasa/fprime #241: Duplicate Enums in Dictionary

Joshua-Anderson, 2020-11-09:
There doesn't seems to be any deduplication logic when adding enums to the deployment's dictionary.

The Ref deployment, for example, has the `Ref::SignalType` enum in the dictionary a half dozen times.
We should investigate what is causing this.

```
  <enums>
    <enum type="CommandDispatcher::error::ErrorResponse">
      <item name="ERR_INVALID_OPCODE" value="0" description="Invalid opcode dispatched"/>
      <item name="ERR_VALIDATION_ERROR" value="1" description="Command failed validation"/>
      <item name="ERR_FORMAT_ERROR" value="2" description="Command failed to deserialize"/>
      <item name="ERR_EXECUTION_ERROR" value="3" description="Command had execution error"/>
      <item name="ERR_BUSY" value="4" description="Component busy"/>
      <item name="ERR_UNEXP" value="5" description="Unexpected response"/>
    </enum>
    <enum type="CommandDispatcher::Status::CmdSerError">
      <item name="ERR_BUFFER_TOO_SMALL" value="0" description="Buffer too small"/>
      <item name="ERR_BUFFER_FORMAT" value="1" description="Buffer wrong format"/>
      <item name="ERR_SIZE_MISMATCH" value="2" description="Buffer size mismatch"/>
      <item name="ERR_TYPE_MISMATCH" value="3" description="Buffer type mismatch"/>
      <item name="ERR_UNEXP_STAT" value="4" description="Unexpected status"/>
    </enum>
    <enum type="PrmDb::stage::PrmWriteError">
      <item name="PRM_WRITE_OPEN" value="0"/>
      <item name="PRM_WRITE_DELIMETER" value="1"/>
      <item name="PRM_WRITE_DELIMETER_SIZE" value="2"/>
      <item name="PRM_WRITE_RECORD_SIZE" value="3"/>
      <item name="PRM_WRITE_RECORD_SIZE_SIZE" value="4"/>
      <item name="PRM_WRITE_PARAMETER_ID" value="5"/>
      <item name="PRM_WRITE_PARAMETER_ID_SIZE" value="6"/>
      <item name="PRM_WRITE_PARAMETER_VALUE" value="7"/>
      <item name="PRM_WRITE_PARAMETER_VALUE_SIZE" value="8"/>
    </enum>
...
    <enum type="Ref::SignalType">
      <item name="TRIANGLE" value="0" description=""/>
      <item name="SQUARE" value="1" description=""/>
      <item name="SINE" value="2" description=""/>
      <item name="NOISE" value="3" description=""/>
    </enum>
    <enum type="Ref::SignalType">
      <item name="TRIANGLE" value="0" description=""/>
      <item name="SQUARE" value="1" description=""/>
      <item name="SINE" value="2" description=""/>
      <item name="NOISE" value="3" description=""/>
    </enum>
...
    <enum type="Ref::SignalType">
      <item name="TRIANGLE" value="0" description=""/>
      <item name="SQUARE" value="1" description=""/>
      <item name="SINE" value="2" description=""/>
      <item name="NOISE" value="3" description=""/>
    </enum>
  </enums>
```

Kemyt, 2021-03-04:
This functionality is present in [gds_dictgen.py](https://github.com/nasa/fprime/blob/0e731d14c161a5c5dc37a2ce2e55009353806247/Autocoders/Python/bin/gds_dictgen.py#L496), but not in codegen.py. A quick fix would be to copy/paste it, since there already seems to be a lot of duplicated code between those two files. Maybe some of the common code could be extracted into a separate file to assure that both scripts are up to date?

billallen256, 2021-03-05:
I would advocate for moving it to a separate file that both could call, plus it would help make `generate_xml_dict()` a bit smaller.

billallen256, 2021-03-05:
Based on [this line](https://github.com/nasa/fprime/blob/0e731d14c161a5c5dc37a2ce2e55009353806247/Autocoders/Python/bin/gds_dictgen.py#L27), I assume it would make the most sense to put the common enum checker in `fprime_ac.utils`?

Joshua-Anderson, 2021-03-05:
@billallen256 Yeah, putting that common deduping logic in utils seems like a good idea to me!

LeStarch, 2021-03-05:
Yes, a common file could be put in `fprime_ac.utils`.

PR #447 (2021-03-07): Move enum duplicates removing to separate file
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**| Autocoders |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**| #241 |
|**_Has Unit Tests (y/n)_**| n |
|**_Builds Without Errors (y/n)_**| y |
|**_Unit Tests Pass (y/n)_**|  |
|**_Documentation Included (y/n)_**| y |

---
## Change Description

Functionality to remove duplicates from list of enums was moved to
separate file to allow both gds_dictgen.py and codegen.py to use it
without duplicating code. Also a few typos were spotted and fixed.

## Rationa [... cut]
closed 2021-03-07 by LeStarch