nasa/fprime #251: Resolve POINTER_CAST and Pointer Serialization

LeStarch, 2020-11-13:
POINTER_CAST may be used for serialization, but it is machine dependent.  We should investigate.

matt392code, 2024-12-27:
[pointer-serialization.txt](https://github.com/user-attachments/files/18264326/pointer-serialization.txt)
Code addresses:
1. Platform Independence:
- Uses uintptr_t for consistent pointer handling
- Implements byte-by-byte serialization for endianness control
- Stores pointer size information for cross-platform validation

2. Safety Features:
- Validates pointer size during deserialization
- Checks alignment requirements
- Provides a framework for runtime pointer validation

3. F Prime Integration:
- Follows F Prime serialization patterns
- Implements proper error handling using SerializeStatus
- Maintains compatibility with existing F Prime components

To implement this solution:
1. Replace direct POINTER_CAST usage with SafePointerSerializer
2. Add validation checks appropriate for your specific use case
3. Define valid memory ranges for pointer validation
4. Consider adding additional metadata for pointer type validation

thomas-bc, 2025-03-03:
@LeStarch is this ticket being addressed by https://github.com/nasa/fprime/pull/3286 ?

LeStarch, 2025-03-03:
Sort-of. This ticket needs to document the expected use case for PointerCastType:

1. Exactly the size of pointer on the system
2. Set to some serializable type
3. Not to leave the system.

LeStarch, 2025-03-03:
I added this as a sub-issue for the parent.  It will now be handled within the context of the work.

PR #3381 (2025-03-13): Adding warning about the use of PlatformPointerCast. Fixes: #251
| | |
|:---|:---|
|**_Related Issue(s)_**|  |
|**_Has Unit Tests (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

Fixes #251
commit eebd2c17 (2025-03-13): Adding warning about the use of PlatformPointerCast. Fixes: #251 (#3381)
closed 2025-03-13 by LeStarch