nasa/fprime #273: seq_gen : Error in example .seq file

jonathanmichel, 2020-11-25:
When using [seq_gen.py](https://github.com/nasa/fprime/blob/devel/Gds/src/fprime_gds/common/tools/seqgen.py) the [cmd_xml_loader](https://github.com/nasa/fprime/blob/devel/Gds/src/fprime_gds/common/loaders/cmd_xml_loader.py) returns "components.mnemonic". If mnemonic is given without component name in .seq file, it will fail ([here](https://github.com/nasa/fprime/blob/22a032e5f98bbe4c91a91b88ec756aa567db7588/Gds/src/fprime_gds/common/tools/seqgen.py#L79)) binary generation when trying to find command in xml dictonary.

https://github.com/nasa/fprime/blob/22a032e5f98bbe4c91a91b88ec756aa567db7588/Gds/examples/simple_sequence.seq#L9

LeStarch, 2020-11-30:
Yes this is a bug and we'll fix it.

Joshua-Anderson, 2020-12-01:
This was fixed in #290 and will be part of the next release.

closed 2020-12-01 by Joshua-Anderson