nasa/fprime #285: `fprime-util` -b Flag

LeStarch, 2020-12-01:
## -b Flag

I would expect the -b flag to allow users to point to a set of build caches and it would choose the "correct" build cache (based on build type). It does not, it want a specific build cache regardless of build type. I'd propose the following:

    1. Add a -d flag to specify the deployment it will be built for
    2. Add a not to help for -b that it is an "advanced feature"
    3. Prevent the specification of --build-type and -b at the same time

Expected use pattern (inverse of above):

```
cd Svc/CmdDispatcher
fprime-util build -d ../Ref
```

Joshua-Anderson, 2020-12-01:
What is the use case for the -b flag? Is it so you can move your build cache to a location of your choosing? Does it point to an individual build cache or a directory holding the set of build caches?

LeStarch, 2020-12-02:
-b is to point at and generate a specific build cache.  It points at the directory of the build cache itself (not the owning directory).

-d would point at the deployment and add a slightly more usable layer to it.

Joshua-Anderson, 2020-12-02:
Why have both `-p` and `-d`? It seems like `-p` can do everything that `-d` can do, right?

LeStarch, 2020-12-02:
They are inverses.  `-p`  Allows you to build a specified path from a WD of a  deployment.  `-d` allows you to build this WD using a specified deployment

LeStarch, 2020-12-10:
We need to test `-p` and `-d`.

LeStarch, 2020-12-10:
To close out:

1. Remove -b flag (work around is to use raw CMake)
2. Test `-p` and `-d` flags work as expected.

PR #315 (2020-12-15): Fix fprime-util deployment flag and remove build cache flag
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**| fprime-util |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**| Resolves #285 |
|**_Has Unit Tests (y/n)_**| n/a |
|**_Builds Without Errors (y/n)_**| y |
|**_Unit Tests Pass (y/n)_**| y |
|**_Documentation Included (y/n)_**| y |

---
## Change Description

- Fixes fprime-util `--deployment` flag
- Removed fprime-util `--build-cache` flag

## Rationale

`--build-cache` is a very advanced option, users are better off using a combination of th [... cut]
closed 2020-12-15 by LeStarch