nasa/fprime #299: Build Broken on Mac OSX

LeStarch, 2020-12-03:
The (framework) build doesn't work on OSX.

LeStarch, 2020-12-03:
```
error: unknown warning option '-Wno-stringop-truncation'; did you mean '-Wno-string-conversion'?
```

LeStarch, 2020-12-03:
Note: the first error is caused by the second.  We should also compare these flags with those in the  Darwin.cmake platform.

Joshua-Anderson, 2020-12-03:
Ah, I forgot to test on Darwin after my fixes. The flags in Darwin.cmake are in addition to the ones in the root cmakelists file. I'll add the `-Wno-stringop-truncation` only for gcc. Do we care about the root cmake project being able to build in old versions of gcc (I.E. gcc 4)? This flag was added recently enough that it's not supported in super old GCC versions, but I can add a flag that tells the compiler to ignore unknown flags so it'll still build. But I think since this build is used only for CI and testing I'm not too worried about supporting old GCC

LeStarch, 2020-12-03:
I am ok with it.  I would also test removing many of those Darwin flags...as why have them, they don't make a lot of sense.

Joshua-Anderson, 2020-12-08:
Resolved by #301

closed 2020-12-08 by Joshua-Anderson