nasa/fprime #322: GDS GUI blank after custom type used in port

Galatis-Giuliano, 2021-01-02:
## Problem Description

I am working on a project which requires an array to be used as the input to a component. We are currently making modification to the MathComponent created in the Math Tutorial to accommodate our project. We attempted to follow the specifications shown at this link to create our .xml files. We are running into issues when trying to run the GDS GUI. We just receive a blank page. We figured out that this issue is caused when changing the port type in the math sender to our custom array. The .xml files are shown below.

```
<component name="MathSender" kind="active" namespace="Ref">
    <import_array_type>Ref/MathTypes/TestArrayAi.xml</import_array_type>
    <import_port_type>Ref/MathPorts/MathOpPortAi.xml</import_port_type>
    <import_port_type>Ref/MathPorts/MathResultPortAi.xml</import_port_type>
    <comment>Component sending a math operation</comment>
    <ports>
        <port name="mathOut" data_type="Ref::MathOp" kind="output">
            <comment>
            Port for sending the math operation
            </comment>
        </port>
        <port name="mathIn" data_type="Ref::MathResult" kind="async_input">
            <comment>
            Port for returning the math result
            </comment>
        </port>
    </ports>
    <commands>
        <command kind="async" opcode="0" mnemonic="MS_DO_MATH"> 
            <comment>
            Do a math operation
            </comment>
            <args>
                <arg name="val1" type="Ref::Test">
                    <comment>The first value</comment>
                </arg>          
                <arg name="val2" type="F32">
                    <comment>The second value</comment>
                </arg>          
                <arg name="operation" type="ENUM">
                    <enum name="MathOp">
                        <item name="ADD"/>
                        <item name="SUBTRACT"/>
                        <item name="MULTIPLY"/>           
                        <item name="DIVIDE"/>           
                    </enum>
                    <comment>The operation to perform</comment>
                </arg>
             </args>
        </command>
    </commands>
    <telemetry>
        <channel id="0" name="MS_VAL1" data_type="Ref::Test">
            <comment>
            The first value
            </comment>
        </channel>
        <channel id="1" name="MS_VAL2" data_type="F32">
            <comment>
            The second value
            </comment>
        </channel>
        <channel id="2" name="MS_OP" data_type="ENUM">
            <enum name="MathOpTlm">
                <item name="ADD_TLM"/>
                <item name="SUB_TLM"/>
                <item name="MULT_TLM"/>           
                <item name="DIV_TLM"/>           
            </enum>
            <comment>
            The operation
            </comment>
        </channel>
        <channel id="3" name="MS_RES" data_type="Ref::Test">
            <comment>
            The result
            </comment>
        </channel>
    </telemetry>
    <events>
        <event id="0" name="MS_COMMAND_RECV" severity="ACTIVITY_LO" format_string="Math Cmd Recvd: %f %f %d">
            <comment>
            Math command received
            </comment>
            <args>
                <arg name="val1" type="Ref::Test">
                    <comment>The val1 argument</comment>
                </arg>          
                <arg name="val2" type="F32">
                    <comment>The val2 argument</comment>
                </arg>          
                <arg name="op" type="ENUM">
                    <comment>The requested operation</comment>
                <enum name="MathOpEv">
                    <item name="ADD_EV"/>
                    <item name="SUB_EV"/>
                    <item name="MULT_EV"/>           
                    <item name="DIV_EV"/>           
                </enum>
                </arg>          
            </args>
        </event>
        <event id="1" name="MS_RESULT" severity="ACTIVITY_HI" format_string = "Math result is %f" >
            <comment>
            Received math result
            </comment>
            <args>
                <arg name="result" type="Ref::Test">
                    <comment>The math result</comment>
                </arg>          
            </args>
        </event>
    </events>
</component>
```

Array definition

```
<array name="Test" namespace="Ref">
  <type>F32</type>
  <size>3</size>
  <format>%f</format>
  <default>
    <value>0.0f</value>
    <value>0.0f</value>
    <value>0.0f</value>
  </default>
</array>
```

When running the GDS we see the following.

[image]

We were wondering if anybody else had run into this problem and if so, what a potential solution might be. Any advice is appreciated. Than you.

LeStarch, 2021-01-04:
Is it possible you could enable "web developer tools" and post the contents of the "Console" tab?  There is likely a JavaScript error or something that needs looking at.

Galatis-Giuliano, 2021-01-06:
In the Firefox console I do get the following errors:

[image]

LeStarch, 2021-01-06:
It looks like you are trying to use a complex type (Array/Serializable) as a command argument.  As far as I know, this isn't supported in the GUI due to the difficulty of preparing/processing the user input form and associated response.

Galatis-Giuliano, 2021-01-07:
Thank you for the heads up. Do you know if this would be supported by the CLI or would our design not be supported by FPrime in general?

LeStarch, 2021-01-10:
I don't think any of our ground code supports commands of complex types at this time.  The embedded code should support it but a custom ground layer would need to be developed to assemble packets from user inputs.

LeStarch, 2021-03-09:
Currently this documents the state of the architecture.  If we need to support complex types in commanding we can open up an enhancement ticket.

closed 2021-03-09 by LeStarch