nasa/fprime #323: "Ref_exe" does not exist.

Ruben1729, 2021-01-06:
| | |
|:---|:---|
|**_Python Version_**|3.6.9|
|**_CMake Version_**|3.10.2|
|**_Commit_**|08b535c|
---
## Problem Description

I was attempting to run `fprime-util generate` but it generates this error

[image]

[image]

I've tried purging and generating again but I get stuck on the same problem.

## How to Reproduce

The following is a script my friend made to setup the project. This will go up to the deployment of the gds for the Ref project.

```bash
#!/bin/bash

# Install prerequisites
echo
echo Installing dependencies ...
echo
sudo apt update
sudo apt-get -y install cmake
sudo apt install python3 python3-pip python3-venv
sudo apt install build-essential
sudo apt install git-all

# Create repository folder
echo
echo Where should the fprime repository be cloned? Specify a valid directory.
echo
read -r fprime_dir
mkdir -p "$fprime_dir"

# Go to specified directory
cd "$fprime_dir"
fprime_dir=$PWD

# Clone nasa/fprime repo into folder
echo
echo Cloning fprime repository ...
echo
git clone https://github.com/nasa/fprime "$fprime_dir"
cd "$fprime_dir/.." || exit

# Create and launch python virtual environment
echo 
echo Launching Python3 virtual environment ...
echo 
python3 -m venv ./fprime-venv
source ./fprime-venv/bin/activate

# Install fprime python packages
echo 
echo Installing fprime python packages ... 
echo 
cd "$fprime_dir" || exit
pip install --upgrade wheel setuptools pip
pip install ./Fw/Python
pip install ./Gds

# Build fprime
echo 
echo Building fprime application ... 
echo 
cd Ref || exit
fprime-util generate
fprime-util build --jobs 32

# Run HTML GUI
echo 
echo Starting HTML GUI ... 
echo 
fprime-gds -g html -r ./build-artifacts
```

My error occurs with the `fprime-util generate` command. 

I'm trying to run this from Windows Subsystem for Linux (WSL). I know this works because my friend has successfully run the script multiple times on his computer.

I have also tried doing it without the script and doing it manually but I encounter the same issue.

## Expected Behavior

Being able to run the following lines of code successfully.

```bash
fprime-util generate
fprime-util build --jobs 32

fprime-gds -g html -r ./build-artifacts
```

LeStarch, 2021-01-06:
It looks like the naming of targets is changing between parts of the system.  When you do an `ls` command inside fprime, do you see `Ref` as a folder or `ref`.  Notice the difference in case here.

Ruben1729, 2021-01-06:
When I do `ls` I get `Ref` 

[image]

I'm not sure whether this affects the outcome or not but since we are talking about the possibility of `ref` being `Ref`, I noticed that I can do cd `ref` and cd `Ref` both yielding the same result but appearing to have different paths. I know that they should be considered the same target, so I don't see how this could be the cause of the problem.

[image]

Additionally, I tried looking for Ref_RecvBuffApp and I got the following

[image]

I did the same for Ref_SignalGen

[image]

I noticed that the folder names on my friend's computer show up with `Ref_` at the beginning instead of `ref_`. I'm wondering if this is a setting in my OS.

Joshua-Anderson, 2021-01-06:
That error is probably related to the binaries not having the execute permission, which is strange.

Are you using WSL1 or WSL2?
Both of these issues (file case sensitivity, execute permissions misisng) seem related to filesystem quirks that are probably caused by WSL.

Ruben1729, 2021-01-06:
I updated to WSL 2 and tried running it again and I'm still running into the same problem. I would have to agree that this is most likely a problem with WSL as the initial issue was simply fixed by using `cd Ref` instead of `cd ref`.  The path was messing up the creation of the folders.

Is there something I can change about the python packages not being executed? I tried changing the permission of the folder, running it in sudo mode, but it still failed with Permission Denied. This was the result of my last attempt.

[image]

Joshua-Anderson, 2021-01-06:
Yeah, the WSL1 probably messed some things up. I'd recommend deleting everything and re-installing fprime.

Ruben1729, 2021-01-07:
Great news! I finally fixed it! I just had to run the commands on my c drive instead of my e drive. I'm guessing I didn't set it up properly when I installed it and somehow the permissions were messed up? Anyway, thanks a lot for the help!

closed 2021-01-07 by Ruben1729