nasa/fprime #325: Warning of XML not valid according to schematron but code still compiles

sfregoso, 2021-01-07:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|   |
---
## Problem Description
When running the 'fprime-util impl' on my component, I get the following warning but the code still compiles 100%.

WARNING: XML file CameraManagerComponentAi.xml is not valid according to schematron <path_hidden_by_me>/Autocoders/Python/schema/default/active_comp_schematron.rng.

## How to Reproduce

1. 
2.
3.

## Expected Behavior

A description of the expected behavior.

billallen256, 2021-02-28:
For some reason an `FprimeXmlException` is not raised in this `elif` block, unlike the `if` block above it:

https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/parsers/XmlComponentParser.py#L1235

LeStarch, 2021-03-05:
This is likely an oversight.

billallen256, 2021-03-07:
I took a stab at this and added the `FprimeXmlException` for schematron, but then some of the tests wouldn't pass any more because they were defined as `active` components, but without an `async_input` port.

In most cases I was able to change the `sync_input` to `async_input`.  However, the thorniest one was `interface1`, which has no ports at all:
https://github.com/nasa/fprime/blob/devel/Autocoders/Python/test/interface1/TestComponentAi.xml

I tried hacking in a single `async_input` port for `interface1`, but then started to get linker errors.

Joshua-Anderson, 2021-03-08:
@billallen256 Ah, you're hitting some of the fun edge cases from lesser-used F' features! For that particular case, have you tried making it a passive component?

billallen256, 2021-03-09:
@Joshua-Anderson I feel like I'm learning this from the inside out.  Good stuff.

@sfregoso I think this should be resolved now that #455 and #453 have been merged in.

closed 2021-03-09 by LeStarch