nasa/fprime #329: Receives ValueError: Can't mix absolute and relative paths when trying to run fprime-util impl

saba-ja, 2021-01-10:
| | |
|:---|:---|
|**_F´ Version_**| 1.5 =< |
|**_Affected Component_**| All components with  SerializableAi.xml |
---
## Problem Description

Received the following error:
```
ValueError: Can't mix absolute and relative paths
```
When attempted to run fprime-util impl for Ref/SignalGen component.

## How to Reproduce

1. cd fprime/Ref/SignalGen
2. fprime-util generate
3. fprime-util impl

## Expected Behavior

The script should have generated the component implementation template files.

saba-ja, 2021-01-10:
## Diagnosis:

This issue happens when a component has *SerializableAi.xml.

Which causes __serializable_obj_list to be an empty list. This causes the following call trace:

https://github.com/nasa/fprime/blob/9ed8a1c2a44bb8ac5ed6395d6f18f3a0c3820904/Autocoders/Python/src/fprime_ac/generators/writers/ComponentWriterBase.py#L356
Line 356:

[image]

https://github.com/nasa/fprime/blob/9ed8a1c2a44bb8ac5ed6395d6f18f3a0c3820904/Autocoders/Python/src/fprime_ac/generators/writers/ComponentWriterBase.py#L796

Line 796

[image]

https://github.com/nasa/fprime/blob/9ed8a1c2a44bb8ac5ed6395d6f18f3a0c3820904/Autocoders/Python/src/fprime_ac/utils/buildroot.py#L37

Lines 44, 55

[image]

As shown in the example below the __xml_filename in the __command_obj_list is a file name:

{'_Serialize__xml_filename': **'SignalPairSerializableAi.xml'**, '_Serialize__name': 'SignalPair', '_Serialize__namespace': 'Ref', '_Serialize__comment': '', '_Serialize__xml_includes_list': [], '_Serialize__c_includes_list': None, '_Serialize__members_list': [('time', 'F32', None, '%f', None), ('value', 'F32', None, '%f', None)], '_Serialize__type_id': '0x5BF628DE'}

The value of the xml_filename is used as “path” in os.path.commonpath, but this is not an absolute path which causes the os.path.commonpath to through “ValueError: Can't mix absolute and relative paths” 
See: [https://docs.python.org/3/library/os.path.html#os.path.commonpath](https://docs.python.org/3/library/os.path.html#os.path.commonpath)

In line 45 of buildroot.py if the value returned by get_build_roots() is an absolute path, and the “path” parameter is a file name (relative path) a Value Error will be thrown.

The next issue is line 55 of get_nearest_build_root(). It will return None as default. As shown in the screenshot below there is an attempt to concatenate the returned value with “/” which will throw a TypeError exception

[image]

saba-ja, 2021-01-10:
A quick fix would be removing line 796 and possibly 797 from initSerialIncludes()
https://github.com/nasa/fprime/blob/9ed8a1c2a44bb8ac5ed6395d6f18f3a0c3820904/Autocoders/Python/src/fprime_ac/generators/writers/ComponentWriterBase.py#L796

Ruben1729, 2021-01-10:
Hey! I'm not sure if this helps but I had a similar issue recently #327

Ruben1729, 2021-01-10:
Hey! I was browsing through the issues and noticed this one! I think this one is related to one I posted a few days ago #327! Hope this helps

saba-ja, 2021-01-10:
Thank you Ruben! Closing the issue since it is duplicate of #327.
This issue is tracked under #328 pull request.

closed 2021-01-10 by saba-ja