nasa/fprime #331: General CMake Template updates

saba-ja, 2021-01-12:
| | |
|:---|:---|
|**_F´ Version_**| Release 1.5.3|
|**_Affected Component_**| all |
---
## Feature Description

**Enhancement 1:**

Since impl is mostly used during development add Verbose flag to /Users/janamian/Documents/workstation_3x/fprime/cmake/target/impl.cmake

${FPRIME_FRAMEWORK_PATH}/Autocoders/Python/bin/implgen.py ${AI_XML} -v

**Enhancement 2:**

include EXPORT files for sourcetrail:
set(CMAKE_EXPORT_COMPILE_COMMANDS ON)

**Enhancement 3:**

Generated UT template file names do not match to CMakeList.txt files in each component
UT templates are currently named:
Tester.cpp
Tester.hpp
MainTester.cpp
Also the ComponentAi.xml file is by default "Svc/BufferManager/BufferManagerComponentAi.xml" instead of actual component Ai.xml

The auto generated CMakeList.txt is the following:
```
#### UTS ###
#set(UT_SOURCE_FILES
#  "${FPRIME_FRAMEWORK_PATH}/Svc/BufferManager/BufferManagerComponentAi.xml"
#  "${CMAKE_CURRENT_LIST_DIR}/test/ut/Tester.cpp"
#  "${CMAKE_CURRENT_LIST_DIR}/test/ut/Main.cpp"
#)
#register_fprime_ut()
```

**Enhancement 4:**

fprime-util impl --ut automatically put generated UT files into test/ut folder
instead of component root folder

**Enhancement 5:**

In the auto generated header comment the copyright still shows 2015 as the last day.

codeflight1, 2022-04-21:
I believe Enhancement 5 is no longer necessary since that header has been removed.

thomas-bc, 2023-11-29:
Closing as all of those are either implemented or outdated requests (e.g. XML no longer used).

closed 2023-11-29 by thomas-bc