nasa/fprime #332: General Documentation update

saba-ja, 2021-01-12:
| | |
|:---|:---|
|**_F´ Version_**| 1.5.3|
|**_Affected Component_**| Documentation  |
---
## Feature Description

**Documentation fix 1:**

In page: 

https://github.com/nasa/fprime/blob/devel/docs/INSTALL.md
 link https://github.com/nasa/fprime/blob/devel/docs/UsersGuide/user/AUTOCOMPLETE.md
 does not work

**Documentation fix 2:**

UT tutorial has outdated info: https://github.com/nasa/fprime/blob/devel/docs/UsersGuide/user/unit-testing.md

**Documentation fix 3:**

Typo in https://github.com/nasa/fprime/blob/devel/docs/UsersGuide/user/port-comp-top.md
A port may specify zero or more arguments of any D´ (supposed to be F`?) data 
type or primitive (int, float, U8, etc.).

**Documentation fix 4:**

Currently the following document
https://github.com/nasa/fprime/blob/update/ground-prototype/docs/Tutorials/MathComponent/Tutorial.md#24132-test-code-implementation
Shows ```fprime-util build –ut``` to run the test.

Add note for ``` fprime-util generate –ut``` and ```fprime-util check``` to clarify running the UT.

**Documentation fix 5:**

https://github.com/nasa/fprime/blob/devel/docs/Tutorials/MathComponent/Tutorial.md#321-component-imports
```
<import_component_type>Svc/PassiveTextLogger/PassiveTextLoggerComponentAi.xml</import_component_type>
```
has changed to 
```
<import_component_type>Svc/PassiveConsoleTextLogger/PassiveTextLoggerComponentAi.xml</import_component_type>
```

**Documentation fix 6:**

https://github.com/nasa/fprime/issues/171 is still not fixed in update/ground-prototype branch

**Documentation fix 7:**

https://github.com/nasa/fprime/blob/devel/docs/Tutorials/MathComponent/Tutorial.md#41-running-the-ground-system

```fprime-gds -d fprime/Ref```

Is -d argument still available? Get argument not found when using -d.

zacharysarah, 2021-02-20:
Mostly fixed by #382; the unit test tutorial may still need an update.

Joshua-Anderson, 2021-02-20:
@saba-ja What outdated info was in the unit testing tutorial?

saba-ja, 2021-02-20:
@Joshua-Anderson 
The followings need to be updated to use fprime-util tool instead of old fprime build tool:

> To build unit tests, go to the component directory (not the test/ut directory) and run ~make ut~ `fprime-util generate --ut` 
 `fprime-util build --ut` to build the unit test ~with code coverage enabled, or run make ut_nocov to build with code coverage disabled~.

> To run unit tests, go to the component directory (not the test/ut directory) and run ~make run_ut~ `fprime-util check --coverage`  to run with coverage enabled ~, or make run_ut_nocov to run with coverage disabled~.

The following seems outdated and redundant:
> ~To generate the analysis, go to the component directory (not the test/ut directory). After building and running the tests as outlined above, run make cov.~ The review the analysis, go to the test/ut directory (not the component director) and review the summary output _gcov.txt files. Next, go to the component directory to review the coverage annotation .hpp.gcov and .cpp.gcov source files.~

I also think it would be helpful to add some information for other available UT options including "leack", "coverage" and "all". See https://github.com/nasa/fprime/blob/5d631af802664719a2de61bf22cf0a1cf3b2dc9f/Fw/Python/src/fprime/fbuild/builder.py#L592-L618

LeStarch, 2021-05-05:
This seems fixed. Closing.

closed 2021-05-05 by LeStarch