nasa/fprime #335: Receives [ERROR] Unknown arguments: -b when running ./docs/doxygen/generate_docs.bash

saba-ja, 2021-01-19:
| | |
|:---|:---|
|**_F´ Version_**| 1.5>=|
|**_Affected Component_**|all   |
---
## Problem Description

Receives [ERROR] Unknown arguments: -b when running:
./docs/doxygen/generate_docs.bash

## How to Reproduce
run the following from fprime directory:
```./docs/doxygen/generate_docs.bash```

## Expected Behavior
Generate doxygen and sphinx documents

saba-ja, 2021-01-19:
## Diagnosis:

The issue is raised from:
docs/doxygen/generate_docs.bash Line 57

[image]

Removing -b argument  does not solve the issue as it gives the following error:

[image]

The new issue is raised from:
Fw/Python/src/fprime/fbuild/builder.py line 258

[image]

The root cause of the problem is Fw/Python/src/fprime/fbuild/builder.py line 196.
[image]

"CMAKE_DEFAULT_BUILD_NAME" adds "build-fprime-automatic-" as prefix to the name for build directory. This causes "docs-build-for-docs" which is defined in "docs/doxygen/generate_docs.bash" line 57 be perceived as "build-fprime-automatic-docs-build-for-docs"

The easiest way to resolve the issue is by changing the cmake folder to build-fprime-automatic-docs (Line 53, 54, 64) and use “docs” as argument to fprime-util build (line 57) of docs/doxygen/generate_docs.bash

[image]

LeStarch, 2021-01-19:
Your suggested fix looks correct.

saba-ja, 2021-01-19:
Fixed in #337. The issue is resolved.

commit 13f32d1e (2021-01-19): update: fixes #335

Failing issue of generate_docs.bash when generating doxygen
is resolved. The script will use a temporary folder name with a prefix
that is used by fprime-util build.
commit accf524f (2021-01-19): Merge pull request #337 from saba-ja/update/ground-prototype

update: fixes #335
closed 2021-01-19 by saba-ja