nasa/fprime #340: This code is using more memory than it needs to.

saba-ja, 2021-01-22:
_Originally posted by @bocchino in https://github.com/nasa/fprime/pull/283#discussion_r533862652_

This code is using more memory than it needs to. The `bufferMem` pointer is stored twice. The other information in the `Fw::Buffer` can be computed when the buffer is allocated. It doesn't need to be stored in the reserved memory area.

https://github.com/nasa/fprime/pull/283#discussion_r533887917

LeStarch, 2021-12-20:
Also see: #341 for the other suggestions

LeStarch, 2021-12-20:
And #342 for the final comments.

matt392code, 2024-12-28:
[buffer-manager.txt](https://github.com/user-attachments/files/18265238/buffer-manager.txt)
Suggested optimized implementation of the BufferManager based on the proposed improvements. This incorporates both the memory efficiency fixes and the LIFO queue approach, as it appears to be the simpler and more maintainable solution.

Key improvements in this implementation:
1. Memory Efficiency:
- Eliminated redundant pointer storage
- Buffer metadata (size, context) computed on allocation
- Contiguous memory allocation for each bin's buffers
- Efficient index-based tracking of available buffers

2. Safety:
- Added bounds checking assertions
- Proper cleanup in destructors
- Buffer ownership verification on deallocation
- Memory alignment handling via placement new

3. Performance:
- O(1) deallocation time
- O(n) allocation time where n is number of bins
- No memory fragmentation within bins
- Efficient LIFO queue implementation using array-based stack

4. Added Features:
- Buffer size verification on allocation
- Manager ID verification on deallocation
- Support for multiple bin sizes
- Clear separation of bin and manager responsibilities

Notable design choices:
1. Used array-based LIFO queue instead of linked lists for:
- Better cache locality
- Simpler implementation
- Reduced memory overhead
- More predictable performance

2. Separated BufferBin class to:
- Encapsulate bin-specific logic
- Make code more maintainable
- Allow for future optimizations per bin
- Simplify testing

closed 2025-12-19 by thomas-bc