nasa/fprime #344: Discussion: Future of XML schema validation in F'

Joshua-Anderson, 2021-01-28:
F' uses relaxNG schemas to validate the structure of XML files.

Every XML file has a `<?xml-mode>` or `<?oxygen>` (we're not very consistent) tag that specifies a relative path to a relaxng schema that can be used for validation.

Some examples:

```
<?xml-model href="../../Autocoders/Python/schema/ISF/component_schema.rng" type="application/xml" schematypens="http://relaxng.org/ns/structure/1.0"?>
<?oxygen RNGSchema="file:../xml/ISF_Type_Schema.rnc" type="compact"?>
```

The issue is that over the years the schema files have been renamed and the component xml files have been copied / pasted / moved around over the years. The result in nearly every one of these `<?xml-mode>` tags points to a non-existent schema file. There isn't anything in the build system that even warns about invalid `<?xml-mode>` tags, and the tools that read the xml files explicitly validate off a known schema based off the xml file name, ignoring the built in `<?xml-mode>`, so this issue has gone undiscovered/ignored for some time.

Since the status quo is broken, we should reconsider how we utilize XML schemas in F'.

I see three options for how we utilize XML schemas going forward.

### Option 1 - Fix and continue using relative paths

We could correct all the existing XML component files to point to the appropriate schema file.

Pros:

- Easy to understand what schema a particular XML file is using.
- Some IDEs can use this schema file to validate XML files while editing

Cons:

- Easy to break: The status quo is a good example, as over time schema files move or files get moved and break the relative import path
- We'd probably have to setup a linter or something to prevent breakages in the future

### Option 2 - Import schema using URL

Instead of importing a schema file via a relative path from the local filesystem, import it using an explicit URL.

Pros:
- Generally considered "best practice" for including a XML schema
- Some IDEs can use this schema file to validate XML files while editing

Cons:
- We'd have to host the xml schemas and version them so that future schema changes don't break pre-existing components on older fprime versions, which is a good amount of work

### Option 3 - Stop including a XML schema

Stop including a schema to validate against in components XML files.

Pros:
- No schema imports means we can't break schema imports
- No maintenance burden and lowest effort path

Cons:
-  Tooling has to explicitly provide a schema validate against: Our tooling already does this because we don't want to trust the schema the user set, since it may not be appropriate for that xml file.
- Breaks any IDE based schema validation

### Overall

I personally would advocate for option #3, since it requires the least amount of long term effort to maintain, can't get outdated, and the only real downside is breaking IDE based validation of XML files. I'm not sure many F' developers use this functionality, or how important it is for them. I place a high value on IDE support, but it may just be because I don't know what I'm missing out on since visual studio code doesn't support relaxng.

What are community member and F' contributors views on this?

timcanham, 2021-01-28:
Since GitHub is now our core repo, can we point at the schemas here?

Joshua-Anderson, 2021-01-28:
We probably want to version the XML schemas separately from F' itself. Since schemas change pretty rarely, needing to update the schema url/version for all 500 xml files for every release would be a pain. Linking to the xml files on github while versioning seperately from fprime would be a little tricky, but we might be able to solve it (maybe a separate set of the git tags for xml schema versions?)

My biggest concern would by link length https://raw.githubusercontent.com/nasa/fprime/NASA-v1.5.3/Autocoders/Python/schema/default/component_schema.rng is pretty long, as compared to something more typical, like http://www.docbook.org/xml/5.0/rng/docbook.rng

gershwinlabs, 2021-02-28:
Creating another repo just for the schemas is also an option.  This would shorten the link length considerably.

Regardless of hosting location, I would advocate for option 2 since it reduces over-reliance on IDE's, and allows more options of automated testing tools and better CI.

billallen256, 2021-02-28:
Moving the schemas to a separate repo would also make it easier to version them independently of F' itself.

Another option for shortening the links is to move the schemas to a top-level directory in the repo.

Joshua-Anderson, 2021-06-01:
Since there's very slim IDE ReactNG schema support and XML modeling will be superseded by F'' YAML based modeling, we've decided to go with the path of stripping out the XML schemas from the core F' schemas (See #663).

Tools interested in using the schemas can manually map object type (Component/Port/Enum/etc) to RelaxNG schemas, similar to how the autocoder works.

Nothing prevents projects using F' from adding XML schema references to their F' objects.

PR #663 (2021-06-02): Remove schema refrence in XML files
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**|  |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**| fixes #344 |
|**_Has Unit Tests (y/n)_**|  |
|**_Builds Without Errors (y/n)_**|  |
|**_Unit Tests Pass (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

Removed all references to xml schemas in F' XML files. 
The autocoder still validates all XML files against their schema.

## Rationale

Our XML files had incorrect schema paths, so they needed to be corrected or removed.
Because o [... cut]
closed 2021-06-02 by LeStarch