nasa/fprime #349: GDS decoding ENUM events incorrectly

saba-ja, 2021-02-04:
| | |
|:---|:---|
|**_F´ Version_**| 1.5> |
|**_Affected Component_**|All   |
---
## Problem Description

Originally reported by @bcmetz 

GDS crashes when string_format has `%d`

The enum definition xml
```
<?xml version="1.0" encoding="UTF-8"?>
<?oxygen RNGSchema="file:../../Autocoders/Python/schema/ISF_Type_Schema.rnc" type="compact"?>
<enum namespace = "Backplane" name="PowerState">
    <comment>Enum of different supported signal types.</comment>
    <item name="OFF" value="0"/>
    <item name="ON" value="1"/>
</enum>
```

arg xml for the event:
```
<event id="0" name="POWERED" severity="ACTIVITY_HI" format_string = "Instrument has been powered %d" >
            <comment>
            Power state of the instrument
            </comment>
            <args>
                <arg type="Backplane::PowerState" name="State"> </arg>
	   </args>
</event>
```
When the bp_C4D.POWER, ON command is sent, the telemetry is pushed correctly:
[image]

But GDS python code crashes:
```
File "fprime_gds/flask/events.py", line 66, in get
    event.template.format_str % tuple([arg.val for arg in event.args]),
TypeError: %d format: a number is required, not str
```

## Expected Behavior

The Event should be printed correctly in the event table in the GDS GUI

saba-ja, 2021-02-04:
The above issue is due to using Python string interpolation (string_format % tuple of args) which enforces type validation. When an unexpected type is given a TypeError exception is thrown.
Gds/src/fprime_gds/flask/events.py#L66

event.template.format_str % tuple([arg.val for arg in event.args])
In the case of above issue the format_string template has integer place holder Instrument has been powered %d while the argument associated with the template is string ON:

[image]

The string interpolation is a common code vulnerability pattern in current FPrime GDS.

saba-ja, 2021-02-04:
Similar to https://github.com/nasa/fprime/issues/343

LeStarch, 2021-03-11:
@saba-ja did we fix this?

closed 2021-05-06 by saba-ja