nasa/fprime #367: Cleaning a subdirectory of a build cache

bocchino, 2021-02-16:
| | |
|:---|:---|
|**_F´ Version_**| 1.5.3 |
|**_Affected Component_**| All |
---
## Feature Description

It would be helpful to have an `fprime-util clean` capability that cleans just a subdirectory of a build cache.

## Rationale

Due to bugs or quirks in the build, the build environment can get itself into a state where it is impossible to proceed without wiping out build artifacts and starting over. Often it is sufficient to wipe out the artifacts in a particular directory, for example a component unit test directory. Wiping out the entire build cache is inconvenient in this case.

Joshua-Anderson, 2021-05-25:
Do you know what situations you were in where CMake was got stuck and was unable to properly rebuild a component after modifications? I've hit a few of those cases in the past but I think we've fixed most of the dependency tree issues now (post 1.5 release).

In theory, cmake should never need to be cleaned and should always rebuild properly after modifying C++ and XML files.

The reason I'm bringing this up is I'm trying to figure out what the use case would be for cleaning just a specific component.
Usually in the cases when cmake gets into a stange state it's broken enough that I'm going to want to recreate the entire build cache anyway.

bocchino, 2021-05-25:
Issues I have encountered in the past include

* Failure to properly re-generate C++ files after modifying XML files.
* Failure to run component unit tests correctly after modifying the component and/or the unit tests.

I think both of these issues have been fixed.

LeStarch, 2021-12-20:
The core errors have been fixed, and it is unclear what cleaning a subdirectory would entail as CMake doesn't have the notion of a local clean.

closed 2021-12-20 by LeStarch

warkuszmonika, 2023-10-25:
> **_F´ Version_**	1.5.3
> **_Affected Component_**	All
> ## Feature Description
> It would be helpful to have an `fprime-util clean` capability that cleans just a subdirectory of a build cache.
>
> ## [...]