nasa/fprime #368: Code coverage analysis causes unit test failures on Mac OS

bocchino, 2021-02-16:
| | |
|:---|:---|
|**_F´ Version_**| 1.5.3 |
|**_Affected Component_**| All |
---
## Problem Description

On Mac OS, the following can occur:

1. Build and run component unit tests successfully. Doing this runs code coverage analysis, which generates .gcda artifacts.
2. Edit the component and/or the unit tests.
3. Build and run the component unit tests again, causing the coverage analysis to be re-run. This time the code coverage analysis gets confused by the old .gcda artifacts, causing it to spew warnings and sometimes fail with a segmentation fault.

A workaround is to manually wipe out the .gcda artifacts in the build cache, but this is inconvenient.

## How to Reproduce

Carry out the steps described above, on any component unit test. Do this on Mac OS. For some reason, the issue arises only on Mac OS and not on Linux.

## Expected Behavior

The code coverage analysis should not spew warnings and should not cause segmentation faults or other apparent test failures.

gunjanmimo, 2021-04-22:
is this problem still open? I am interested to take this one.

bocchino, 2021-04-22:
Yes, this issue is still open.

Joshua-Anderson, 2021-04-22:
I think the solution is likely setting up cmake to clear the leftover `gcda` files from previous runs before re-running the test suite with coverage on.

PR #551 (2021-05-14): Always cleanup coverage gcda files before executing unit tests
| | |
|:---|:---|
|**_Originating Project/Creator_**| |
|**_Affected Component_**|  cmake |
|**_Affected Architectures(s)_**|  |
|**_Related Issue(s)_**| fixes  #368  |
|**_Has Unit Tests (y/n)_**| y  |
|**_Builds Without Errors (y/n)_**| y |
|**_Unit Tests Pass (y/n)_**|  |
|**_Documentation Included (y/n)_**|  |

---
## Change Description

gcda files are generated per object file when executing a binary with coverage enabled. Make sure all existing coverage files are removed before re-running unit test executables

## Rationale

 [... cut]
closed 2021-05-14 by LeStarch