nasa/fprime #369: Autocoder does not automatically include CmdPort

saba-ja, 2021-02-18:
| | |
|:---|:---|
|**_F´ Version_**|>1.5 |
|**_Affected Component_**| Autocoder  |
---
## Problem Description

When *ComponentAi.xml only contains parameter and not any command the autocoder does not include
Fw/Cmd/CmdPortAc.hpp which causes the following error:

[image]

## How to Reproduce

1. Use a ComponentAi.xml with parameters and no command
2. generate and build the component

## Expected Behavior

Include CmdPortAc.hpp automatically when parameters are present without command in  *ComponentAi.xml

saba-ja, 2021-02-18:
# Workaround

Manually import port type <import_port_type>Fw/Cmd/CmdPortAi.xml</import_port_type> to *ComponentAi.xml

LeStarch, 2021-03-11:
@saba-ja do you have a fix for this?

saba-ja, 2021-03-20:
The root cause of this problem is in the line 25 of the following:
https://github.com/nasa/fprime/blob/e1624fd8973284aec3a1c08c26732349620c2a6d/cmake/support/parser/ai_parser.py#L15-L29

On line 25 `Fw_Cmd` is not included as one of the dependencies of parameters. This causes the cmake to not copy required Cmd header files despite autocoder implicitly generating Cmd related handlers based on the following:
https://github.com/nasa/fprime/blob/e1624fd8973284aec3a1c08c26732349620c2a6d/Autocoders/Python/src/fprime_ac/parsers/XmlComponentParser.py#L993-L994
https://github.com/nasa/fprime/blob/e1624fd8973284aec3a1c08c26732349620c2a6d/Autocoders/Python/src/fprime_ac/parsers/XmlComponentParser.py#L1028-L1029

I was able to reproduce the error by creating a component with parameters only and running `fprime-util build --ut`. The error is shown below:
[image]

The error was resolved by updating line 25 of fprime/cmake/support/parser/ai_parser.py to
```
"parameters": ["Fw_Cfg", "Fw_Types", "Fw_Prm", "Fw_Cmd"], 
```

The successful build is shown below:
[image]
[image]

closed 2021-03-24 by saba-ja