nasa/fprime #405: Publish gds and fw packages to PyPI

gershwinlabs, 2021-02-22:
| | |
|:---|:---|
|**_F´ Version_**|current|
|**_Affected Component_**|Gds and Fw/python|
---
## Feature Description

Publish `Gds` and `Fw/Python` as `fprime-gds` and `fprime-fw` packages on PyPI.

## Rationale

- Easier installation for users since they can just use `python3 -m pip install fprime-gds`.
- `fprime-gds` package will be able to list `fprime-fw` as a dependency.
- Actual releases.  Users will be using versions of the packages released on PyPI, not random clones from GitHub.

EverLookNeverSee, 2021-04-19:
@billallen256 
I would like to work on this issue.

Joshua-Anderson, 2021-05-05:
Right now fprime-util and fprime-gds are pretty tightly tied to the version of F' you're using. Because different F' deployments/projects could use different versions of the F' framework, we recommend creating a venv for each deployment and then directly installing the two packages from the embedded fprime framework submodule. I guess you could install the specific versions you require from pypi instead, but that doesn't seem like much of an improvement.

Long term there has been some discussion about possibly breaking fprime-util and fprime-gds into their own repositories and versioning them separately, at which point I think publishing them to pypi would make sense, since you would then be able to update them separately from the F' framework.

EverLookNeverSee, 2021-05-06:
I tried to upload version 1.5.3 to test pipy but it returned me error below:
HTTPError: 400 Bad Request from https://test.pypi.org/legacy/
Invalid value for classifiers. Error: Classifier 'License :: OSI Approved :: Apache 2.0' is not a valid classifier.

BastianZim, 2021-05-06:
According to https://pypi.org/classifiers/ the only classifier with Apache is `License :: OSI Approved :: Apache Software License` but that is something that should probably be definitively answered by the maintainers.

billallen256, 2021-05-11:
As @Joshua-Anderson pointed out, I put in this ticket before I realized how tightly coupled everything was.  Decoupling things seems like a good direction to move though, as it would help remove some barriers to entry, and allow things to be more idiomatic.

EverLookNeverSee, 2021-05-20:
@LeStarch Could you please give me some hints in order to solve this issue?

LeStarch, 2021-05-20:
Given the complexity of this task, I have begun to start and split the packages apart and prepare for this.  Before this can be done officially, we need to merge:

#593, #594, #574

closed 2021-05-28 by LeStarch