nasa/fprime #419: fprime-util impl autocoder does not use return type

metzdigital, 2021-02-25:
The return type used in the interface definition is ignored when using `fprime-util impl`, the autocoder return type is hard coded to always be `void`

[image]

After manually replacing `void` with `Backplane::CardConfigReturn` I could get this to compile. 

Here is my interface definition:
```
<interface name="CardConfig" namespace="Backplane">
    <comment>
        Port to grab card configuration details used by the backplane manager at initialization
    </comment>
    <import_serializable_type>Backplane/Common/CardConfigReturnSerializableAi.xml</import_serializable_type>
    <args>
        <!-- No args only the return is important -->
    </args>
    <return type="Backplane::CardConfigReturn" pass_by="value"> </return>
</interface>
```

saba-ja, 2021-02-26:
## Root cause:

The problem is from line 61

https://github.com/nasa/fprime/blob/598f850b0784735acc9a8b6e367a9cd5acdd0972/Autocoders/Python/src/fprime_ac/generators/templates/impl/hpp.tmpl#L53-L64

and 

https://github.com/nasa/fprime/blob/598f850b0784735acc9a8b6e367a9cd5acdd0972/Autocoders/Python/src/fprime_ac/generators/templates/impl/cpp.tmpl#L55-L70

The template always adds `void` for user-defined input port types.

## Fix:

To resolve the issue the following can be done:

In fprime/Autocoders/Python/src/fprime_ac/generators/templates/impl/hpp.tmpl
```c++
  #for $instance, $type, $sync, $priority, $full, $role, $max_num in $typed_user_input_ports:
      //! Handler implementation for $instance
      //!
      #set $return_type = $port_return_type_strs[$instance]
      ${return_type}${instance}_handler(
$emit_port_params([ $param_portNum ] + $port_params[$instance])
      );
```

In fprime/Autocoders/Python/src/fprime_ac/generators/templates/impl/cpp.tmp
```c++
  #for $instance, $type, $sync, $priority, $full, $role, $max_num in $typed_user_input_ports:
  #set $return_type = $port_return_type_strs[$instance]
  ${return_type}${name}ComponentImpl ::
    ${instance}_handler(
$emit_port_params([ $param_portNum ] + $port_params[instance])
    )
  {
  #if $return_type[:4] == "void":    
    // TODO
  #else:
    // TODO return
  #end if
  }
```

## Output after fix:

hpp before:
[image]

hpp after:
[image]

cpp before:
[image]

cpp after:
[image]

## Bonus fix:

`user-defined serial input ports` is also always `void`:

https://github.com/nasa/fprime/blob/598f850b0784735acc9a8b6e367a9cd5acdd0972/Autocoders/Python/src/fprime_ac/generators/templates/impl/hpp.tmpl#L69-L81

https://github.com/nasa/fprime/blob/598f850b0784735acc9a8b6e367a9cd5acdd0972/Autocoders/Python/src/fprime_ac/generators/templates/impl/cpp.tmpl#L72-L84

The same approach can be used to fix the above as well.

LeStarch, 2021-02-26:
@saba-ja you should post a PR to fix the return type!    Serial ports shouldn't return anything...i don't think...so likely we only need the first fix.

closed 2021-03-04 by LeStarch