nasa/fprime #422: Drop Python 2 support

gershwinlabs, 2021-02-27:
| | |
|:---|:---|
|**_F´ Version_**| current |
|**_Affected Component_**| Autocoder |
---
## Feature Description

Remove work-arounds in the code that are there to support Python 2.

## Rationale

There are some parts of the Autocoder that have work-arounds to support Python 2 and Python 3.  Since Python 2 reached end-of-life at the beginning of 2020, support should be dropped so these work-arounds can be removed.

LeStarch, 2021-03-04:
We have formally dropped Python 2 support, however; as you have seen we haven't removed all work around.

LeStarch, 2021-05-05:
We merged the PR here.  I am going to close this.  We can reopen if more are found.

closed 2021-05-25 by Joshua-Anderson