nasa/fprime #425: MathComponent Tutorial, two minor issues with the Tutorial instructions

jmstadt, 2021-03-01:
In the following Tutorial https://github.com/nasa/fprime/blob/devel/docs/Tutorials/MathComponent/Tutorial.md#2421-component-implementation (which as great to do, thanks for sharing), two minor issues:
In https://github.com/nasa/fprime/blob/devel/docs/Tutorials/MathComponent/Tutorial.md#2421-component-implementation where the code says:
Add the stub files to CMakeLists.txt:
`set(SOURCE_FILES
  "${CMAKE_CURRENT_LIST_DIR}/MathReceiverComponent.xml"
  "${CMAKE_CURRENT_LIST_DIR}/MathReceiverComponentImpl.cpp"
)
register_fprime_module()`
It should state:
`set(SOURCE_FILES
  "${CMAKE_CURRENT_LIST_DIR}/MathReceiverComponent**Ai**.xml"
  "${CMAKE_CURRENT_LIST_DIR}/MathReceiverComponentImpl.cpp"
)
register_fprime_module()`

Then in section https://github.com/nasa/fprime/blob/devel/docs/Tutorials/MathComponent/Tutorial.md#24224-event-throttling
Where it says to add to Tester.cpp:
`  void Tester ::
    testThrottle(void) 
  {`

The following also needs to be added to Tester.hpp:
`    public:

      // ----------------------------------------------------------------------
      // Tests
      // ----------------------------------------------------------------------

      //! Test operation command
      //!
      .........
      void testThrottle(void);`

LeStarch, 2021-03-04:
Good catches!  Would you like to make a PR?  Otherwise I'll do it as part of the next documentation updates.

jmstadt, 2021-03-04:
Thank you!  No on the PR, I am sure regular document updates are fine.  It was a great tutorial.  Thank you for posting it.  I am not too good at c++ and those minor issues probably helped me learn some more.  Thank you again.

LeStarch, 2021-05-05:
In progress.

closed 2021-05-06 by LeStarch