nasa/fprime #430: A stack-buffer-overflow error in FileDownlink.cpp line 248

zouyonghao, 2021-03-03:
| | |
|:---|:---|
|**_F´ Version_**| 1.5 |
|**_Affected Component_**|  Svc/FileDownlink |
---
## Problem Description

A stack-buffer-overflow error in [FileDownlink.cpp line 248](https://github.com/nasa/fprime/blob/ada6e424b813e6434284bbde9f785656a61c8539/Svc/FileDownlink/FileDownlink.cpp#L248)

```
zyh@virtual:~/fprime$ ./build-fprime-automatic-native-ut/bin/Linux/Svc_FileDownlink_ut_exe 
[==========] Running 7 tests from 1 test suite.
[----------] Global test environment set-up.
[----------] 7 tests from FileDownlink
[ RUN      ] FileDownlink.Downlink
=================================================================
==16373==ERROR: AddressSanitizer: stack-buffer-overflow on address 0x7ffcbb2f32d0 at pc 0x0000004ea412 bp 0x7ffcbb2f2c30 sp 0x7ffcbb2f23e0
READ of size 101 at 0x7ffcbb2f32d0 thread T0
    #0 0x4ea411 in __asan_memcpy (/home/zyh/fprime/build-fprime-automatic-native-ut/bin/Linux/Svc_FileDownlink_ut_exe+0x4ea411)
    #1 0x62b297 in Svc::FileDownlink::SendFile_cmdHandler(unsigned int, unsigned int, Fw::CmdStringArg const&, Fw::CmdStringArg const&) /home/zyh/fprime/Svc/FileDownlink/FileDownlink.cpp:248:5
    #2 0x677a82 in Svc::FileDownlinkComponentBase::doDispatch() /home/zyh/fprime/build-fprime-automatic-native-ut/F-Prime/Svc/FileDownlink/FileDownlinkComponentAc.cpp:2811:15
    #3 0x52c79b in Svc::Tester::sendFile(char const*, char const*, Fw::CommandResponse) /home/zyh/fprime/Svc/FileDownlink/test/ut/Tester.cpp:529:21
    #4 0x52a126 in Svc::Tester::downlink() /home/zyh/fprime/Svc/FileDownlink/test/ut/Tester.cpp:68:11
    #5 0x56f302 in FileDownlink_Downlink_Test::TestBody() /home/zyh/fprime/Svc/FileDownlink/test/ut/Main.cpp:9:10
    #6 0x833db0 in void testing::internal::HandleSehExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/zyh/fprime/gtest/googletest-src/googletest/src/gtest.cc:2433:10
    #7 0x7fc5d9 in void testing::internal::HandleExceptionsInMethodIfSupported<testing::Test, void>(testing::Test*, void (testing::Test::*)(), char const*) /home/zyh/fprime/gtest/googletest-src/googletest/src/gtest.cc:2469:14
    #8 0x766039 in testing::Test::Run() /home/zyh/fprime/gtest/googletest-src/googletest/src/gtest.cc:2508:5
    #9 0x767fc4 in testing::TestInfo::Run() /home/zyh/fprime/gtest/googletest-src/googletest/src/gtest.cc:2684:11
    #10 0x768df4 in testing::TestSuite::Run() /home/zyh/fprime/gtest/googletest-src/googletest/src/gtest.cc:2816:28
    #11 0x787a0a in testing::internal::UnitTestImpl::RunAllTests() /home/zyh/fprime/gtest/googletest-src/googletest/src/gtest.cc:5338:44
    #12 0x83bca0 in bool testing::internal::HandleSehExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /home/zyh/fprime/gtest/googletest-src/googletest/src/gtest.cc:2433:10
    #13 0x8025b9 in bool testing::internal::HandleExceptionsInMethodIfSupported<testing::internal::UnitTestImpl, bool>(testing::internal::UnitTestImpl*, bool (testing::internal::UnitTestImpl::*)(), char const*) /home/zyh/fprime/gtest/googletest-src/googletest/src/gtest.cc:2469:14
    #14 0x787026 in testing::UnitTest::Run() /home/zyh/fprime/gtest/googletest-src/googletest/src/gtest.cc:4925:10
    #15 0x57627e in RUN_ALL_TESTS() /home/zyh/fprime/gtest/googletest-src/googletest/include/gtest/gtest.h:2473:46
    #16 0x571eaf in main /home/zyh/fprime/Svc/FileDownlink/test/ut/Main.cpp:44:10
    #17 0x7ff0c0f95bf6 in __libc_start_main /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:310
    #18 0x42b699 in _start (/home/zyh/fprime/build-fprime-automatic-native-ut/bin/Linux/Svc_FileDownlink_ut_exe+0x42b699)

Address 0x7ffcbb2f32d0 is located in stack of thread T0 at offset 720 in frame
    #0 0x6743ef in Svc::FileDownlinkComponentBase::doDispatch() /home/zyh/fprime/build-fprime-automatic-native-ut/F-Prime/Svc/FileDownlink/FileDownlinkComponentAc.cpp:2642

  This frame has 22 object(s):
    [32, 192) 'msg' (line 2643)
    [256, 260) 'priority' (line 2644)
    [272, 276) 'desMsg' (line 2655)
    [288, 292) 'portNum' (line 2668)
    [304, 308) 'context' (line 2681)
    [320, 376) 'fwBuffer' (line 2702)
    [416, 420) 'key' (line 2723)
    [432, 436) 'opCode' (line 2743)
    [448, 452) 'cmdSeq' (line 2751)
    [464, 600) 'args' (line 2759)
    [672, 720) 'sourceFileName' (line 2770)
    [752, 800) 'destFileName' (line 2785) <== Memory access at offset 720 partially underflows this variable
    [832, 836) 'opCode55' (line 2824)
    [848, 852) 'cmdSeq60' (line 2832)
    [864, 1000) 'args65' (line 2840)
    [1072, 1076) 'opCode77' (line 2871)
    [1088, 1092) 'cmdSeq82' (line 2879)
    [1104, 1240) 'args87' (line 2887)
    [1312, 1360) 'sourceFileName92' (line 2898)
    [1392, 1440) 'destFileName98' (line 2913)
    [1472, 1476) 'startOffset' (line 2928)
    [1488, 1492) 'length' (line 2943)
HINT: this may be a false positive if your program uses some custom stack unwind mechanism or swapcontext
      (longjmp and C++ exceptions *are* supported)
SUMMARY: AddressSanitizer: stack-buffer-overflow (/home/zyh/fprime/build-fprime-automatic-native-ut/bin/Linux/Svc_FileDownlink_ut_exe+0x4ea411) in __asan_memcpy
Shadow bytes around the buggy address:
  0x100017656600: f1 f1 f1 f1 00 00 00 00 00 00 00 00 00 00 00 00
  0x100017656610: 00 00 00 00 00 00 00 00 f2 f2 f2 f2 f2 f2 f2 f2
  0x100017656620: 04 f2 04 f2 04 f2 f8 f2 f8 f8 f8 f8 f8 f8 f8 f2
  0x100017656630: f2 f2 f2 f2 f8 f2 04 f2 04 f2 00 00 00 00 00 00
  0x100017656640: 00 00 00 00 00 00 00 00 00 00 00 f2 f2 f2 f2 f2
=>0x100017656650: f2 f2 f2 f2 00 00 00 00 00 00[f2]f2 f2 f2 00 00
  0x100017656660: 00 00 00 00 f2 f2 f2 f2 f8 f2 f8 f2 f8 f8 f8 f8
  0x100017656670: f8 f8 f8 f8 f8 f8 f8 f8 f8 f8 f8 f8 f8 f2 f2 f2
  0x100017656680: f2 f2 f2 f2 f2 f2 f8 f2 f8 f2 f8 f8 f8 f8 f8 f8
  0x100017656690: f8 f8 f8 f8 f8 f8 f8 f8 f8 f8 f8 f2 f2 f2 f2 f2
  0x1000176566a0: f2 f2 f2 f2 f8 f8 f8 f8 f8 f8 f2 f2 f2 f2 f8 f8
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==16373==ABORTING
```

## How to Reproduce

1. Compile unit test with ASAN
2. Run build-fprime-automatic-native-ut/bin/Linux/Svc_FileDownlink_ut_exe
3. Program abort

## Expected Behavior

Run normally with ASAN.

Joshua-Anderson, 2021-03-03:
Good find! Those memcpy commands should be replaced with the a call to `Fw::StringUtils::string_copy`.

Joshua-Anderson, 2021-03-03:
Fixed in #435

closed 2021-03-03 by Joshua-Anderson