nasa/fprime #431: A global-buffer-overflow error in BufferQueueTest.cpp line 126

zouyonghao, 2021-03-03:
| | |
|:---|:---|
|**_F´ Version_**|1.5 |
|**_Affected Component_**| Os/Pthreads |
---
## Problem Description

A global-buffer-overflow error in [BufferQueueTest.cpp line 126](https://github.com/nasa/fprime/blob/ada6e424b813e6434284bbde9f785656a61c8539/Os/Pthreads/test/ut/BufferQueueTest.cpp#L126)

```
zyh@virtual:~/fprime$ ./build-fprime-automatic-native-ut/bin/Linux/Os_pthreads
Creating queue.
Test empty queue...
Passed.
Test full queue...
Pushing 0.
Pushing 1.
Pushing 2.
Pushing 3.
Pushing 4.
Passed.
Test weird size...
Passed.
Test pop...
Popping 5.
Popping 4.
Popping 3.
Popping 2.
Popping 1.
Passed.
Test priorities...
Pushing 'hello' at priority 9.
Pushing 'how are you' at priority 4.
Pushing 'pretty good' at priority 100.
Pushing 'cosmic bro' at priority 4.
Pushing 'kthxbye' at priority 9.
Popped 'pretty good' at priority 100. Expected 'pretty good' at priority 100.
Popped 'hello' at priority 9. Expected 'hello' at priority 9.
=================================================================
==17078==ERROR: AddressSanitizer: global-buffer-overflow on address 0x000000550bc6 at pc 0x0000004b3f6b bp 0x7ffd804c4c90 sp 0x7ffd804c4440
READ of size 8 at 0x000000550bc6 thread T0
    #0 0x4b3f6a in __interceptor_memcmp.part.283 (/home/zyh/fprime/build-fprime-automatic-native-ut/bin/Linux/Os_pthreads+0x4b3f6a)
    #1 0x51d718 in main /home/zyh/fprime/Os/Pthreads/test/ut/BufferQueueTest.cpp:126:5
    #2 0x7f976cdadbf6 in __libc_start_main /build/glibc-S9d2JN/glibc-2.27/csu/../csu/libc-start.c:310
    #3 0x41ab89 in _start (/home/zyh/fprime/build-fprime-automatic-native-ut/bin/Linux/Os_pthreads+0x41ab89)

0x000000550bc6 is located 58 bytes to the left of global variable '<string literal>' defined in '/home/zyh/fprime/Os/Pthreads/test/ut/BufferQueueTest.cpp:98:43' (0x550c00) of size 12
  '<string literal>' is ascii string 'how are you'
0x000000550bc6 is located 0 bytes to the right of global variable '<string literal>' defined in '/home/zyh/fprime/Os/Pthreads/test/ut/BufferQueueTest.cpp:98:34' (0x550bc0) of size 6
  '<string literal>' is ascii string 'hello'
SUMMARY: AddressSanitizer: global-buffer-overflow (/home/zyh/fprime/build-fprime-automatic-native-ut/bin/Linux/Os_pthreads+0x4b3f6a) in __interceptor_memcmp.part.283
Shadow bytes around the buggy address:
  0x0000800a2120: f9 f9 f9 f9 00 00 00 00 00 00 00 01 f9 f9 f9 f9
  0x0000800a2130: 00 00 05 f9 f9 f9 f9 f9 00 01 f9 f9 f9 f9 f9 f9
  0x0000800a2140: 00 00 04 f9 f9 f9 f9 f9 00 05 f9 f9 f9 f9 f9 f9
  0x0000800a2150: 00 00 04 f9 f9 f9 f9 f9 00 05 f9 f9 f9 f9 f9 f9
  0x0000800a2160: 00 05 f9 f9 f9 f9 f9 f9 00 00 04 f9 f9 f9 f9 f9
=>0x0000800a2170: 00 00 04 f9 f9 f9 f9 f9[06]f9 f9 f9 f9 f9 f9 f9
  0x0000800a2180: 00 04 f9 f9 f9 f9 f9 f9 00 04 f9 f9 f9 f9 f9 f9
  0x0000800a2190: 00 03 f9 f9 f9 f9 f9 f9 00 f9 f9 f9 f9 f9 f9 f9
  0x0000800a21a0: 00 00 00 06 f9 f9 f9 f9 00 00 04 f9 f9 f9 f9 f9
  0x0000800a21b0: 00 00 00 00 00 00 00 03 f9 f9 f9 f9 00 04 f9 f9
  0x0000800a21c0: f9 f9 f9 f9 00 00 00 00 00 00 00 00 00 00 00 00
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==17078==ABORTING
```

## How to Reproduce

1. Compile unit test with ASAN
2. Run build-fprime-automatic-native-ut/bin/Linux/Os_pthreads
3. Program abort

## Expected Behavior

Run normally with ASAN.

jojavihe, 2021-03-03:
Excelente.excelente

LeStarch, 2021-03-07:
Good catch.  Do note: this is  in a unit test.

LeStarch, 2021-03-07:
I think this will be fixed by #545.  It corrects a number of issues with that test.

closed 2021-03-09 by LeStarch