nasa/fprime #433: Insufficient error handling

zouyonghao, 2021-03-03:
| | |
|:---|:---|
|**_F´ Version_**| 1.5 |
|**_Affected Component_**| Os |
---
## Feature Description

We found a lot of assertions about pthread functions in [Queue.cpp](https://github.com/nasa/fprime/blob/devel/Os/Posix/Queue.cpp) and [Mutex.cpp](https://github.com/nasa/fprime/blob/devel/Os/Posix/Mutex.cpp).
There assertions often follow some comments that claim this situation is dangerous.
Besides, these pthread function like `pthread_cond_init` could fail in real world deployments.
So there should be some error handling code about these functions.

Joshua-Anderson, 2021-03-03:
`pthread_cond_init` can fail due to lack of system memory or a violating function preconditions (programmer error), both of which are tricky to safely handle. The way F' is setup, FW_ASSERT assertions will trigger a FATAL error, which projects can choose to handle at the deployment level by implementing a FatalHandler component. For linux programs the [default handler](https://github.com/nasa/fprime/blob/devel/Svc/FatalHandler/FatalHandlerComponentLinuxImpl.cpp) will log the assert and raise a SIGSEV so that a core dump will be created and the program can be restarted.

zouyonghao, 2021-03-04:
> `pthread_cond_init` can fail due to lack of system memory or a violating function call preconditions (programmer error), both of which are tricky to safely handle. The way F' is setup, FW_ASSERT assertions will trigger a FATAL error, which projects can choose to handle at the deployment level by implementing a FatalHandler component. For linux programs the [default handler](https://github.com/nasa/fprime/blob/devel/Svc/FatalHandler/FatalHandlerComponentLinuxImpl.cpp) will log the assert and raise a SIGSEV so that a core dump will be created and the program can be restarted.

Got it, thanks!

closed 2021-03-04 by zouyonghao