nasa/fprime #436: Is this software actually being used?

robertramey, 2021-03-04:
From the information in this repository, I gather that that this software is being or has been used on actual space flights.  Is this really true?  If that's the case how can there be so many bugs/issues with this software?  This isn't adding up for me.

developerfred, 2021-03-05:
always have bugs in production :)

robertramey, 2021-03-05:
Really?  Even for software destined for Mars?  Honestly, this is impossible for me to believe

JonJakova, 2021-03-05:
Software on which people's life depends on have bugs in them. The Linux kernel on which the world's internet depends on have bugs.
What makes it so hard to believe that this given use case would be bug free?

robertramey, 2021-03-05:
"What makes it so hard to believe that this given use case would be bug free?"

The potential cost of a bug in this environment is orders of magnitude larger than the cost of of making the software bug free.  There are well known documented cases where these kinds of mistakes in this environment have cost hundreds of millions of dollars. It makes no economic sense to use software with this many bugs for this environment.

As I said, it's unbelievable to me that anyone would find this situation acceptable.

LeStarch, 2021-03-05:
This is neither a bug nor an enhancement.  Discussions in general should be done on the mailing list. Thus I am closing this ticket.

closed 2021-03-05 by LeStarch