nasa/fprime #438: Any Recommendations for Radiation Tolerant Programming or applications

TheSpiffiest, 2021-03-04:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Does any part of this software do anything specific to provide radiation tolerant configuration or error checking?

## Rationale

Would like to know if such coding is needed or what the effects would be. Was radiation protection a concern in this code base, or is it not a concern of missions as short as Ingenuity.

billallen256, 2021-03-04:
My understanding is that Ingenuity has a normal CPU, but a radiation-tolerant FPGA that's checking for issues in the processing.

Joshua-Anderson, 2021-03-04:
Here's a [stack overflow post](https://space.stackexchange.com/questions/32999/how-are-cubesat-electronics-protected-against-the-effects-of-radiation-in-leo?rq=1) that might be a good starting place. The answer is very mission and hardware dependent, but usually a hardware watchdog to reboot the computer when the software becomes unresponsive and redundant copies of the software to protect against memory corruption are common starting places.

closed 2021-03-04 by Joshua-Anderson