nasa/fprime #448: File Downlink Not Downlinking Files Correctly

LeStarch, 2021-03-07:
| | |
|:---|:---|
|**_F´ Version_**| devel |
|**_Affected Component_**| FileDownlink  |
---
## Problem Description

FileDownlink isn't downlinking identical files. FileUplink seems to work.  An integration test should be added to test basic up/down behavior.

## How to Reproduce

1. Create a 1MB+ file
2. Use FileDownlink to downlink the file
3. `diff source downlinked`

## Expected Behavior

Identical files.

abcouwer-jpl, 2021-04-12:
I was trying to do file downlink with a topology using the socketIpDriver as the communication wire, and was running into issues where my downliked files, bitmap images, had black sections (all 0s). Looking at the downlink log, it appeared that there were missing data packets. I switched to TCP by calling 

`socketIpDriver.configure(hostname, port_number, 0 /* TCP*/);`

after socketIpDriver.startSocketTask() in the Topology's ConstructApp(), and reliably received images in GDS.

Note that socketIpDriver.startSocketTask() itself calls configure(), defaulting to UDP. Maybe that could be better, maybe startSocketTask() could have a UDP argument.

I don't know the communications wire or the way in which the downlinked files were not identical for @LeStarch , but maybe this is helpful.

abcouwer-jpl, 2021-04-13:
I switched back to udp, and watched for udp loss with

`watch -n 1 -d 'cat /proc/net/udp'`

based on https://serverfault.com/questions/214520/how-to-measure-and-minimize-udp-packet-loss .

When I did the downlink twice, I saw indication of 20 and 29 packets dropping each time, for a total of 49. 0xC351 = 50001, the ip port I was using.

```
  sl  local_address rem_address   st tx_queue rx_queue tr tm->when retrnsmt   uid  timeout inode ref pointer drops             
  149: 00000000:0277 00000000:0000 07 00000000:00000000 00:00000000 00000000     0        0 52486049 2 0000000000000000 0      
  367: 00000000:C351 00000000:0000 07 00000000:00000000 00:00000000 00000000  1000        0 52561407 2 0000000000000000 49     
  416: 00000000:C382 00000000:0000 07 00000000:00000000 00:00000000 00000000  1000        0 52565120 2 0000000000000000 0      
  ...

```

So downlink may be sending udp packets too quickly and overflowing a buffer? F' may be sending lots of small packets (small buffers), overwhelming a network buffer by number of packets, but perhaps not total data size. This is consistent with how the beginnings of my downlinked files were always fine.

Joshua-Anderson, 2021-05-10:
@abcouwer-jpl  I did some investigation and it seems like you're dead on. We're sending file packets as quickly as we can dump them to a UDP socket, and I was seeing UDP send drops as well, which seems to indicate that we're overflowing the UDP send buffer for the program.

I was surprised that there doesn't seem to be any way to configure `send()` to block until there's space in the send buffer or even a good way to determine if the send buffer is full.  

The reason why we currently overflow the UDP send buffer is that FileDownlink file packets are currently sized to fit within a ComLogger buffer (`FILEDOWNLINK_INTERNAL_BUFFER_SIZE`), leaving only ~100 bytes for file contents per packet. This results in thousands of packets for even a modestly sized 1 MB file. Now that FileDownlink packets are no longer stored within ComLogger packets but passed via buffers, we can increase their size to the size of a radio packet. I suspect fewer, larger file packets will be less likely to overwhelm the UDP send buffer, and will hopefully resolve the UDP packet drops.

In general though we should probably switch to TCP and avoid the issue entirely for our reference applications. Since our reference GDS doesn't have the ability to automatically re-request (or even detect) dropped file packets, any UDP packet drops are going to silently corrupt files.

LeStarch, 2021-05-14:
I am prepared to close this issue.  It is fixed by moving to a TCP ground link, which will be done as part of #540.  We could increase the buffer size, but for now TCP fixes the problem.

closed 2021-05-14 by LeStarch