nasa/fprime #449: ActiveTextLoggerComponent does not have async_input port

billallen256, 2021-03-07:
| | |
|:---|:---|
|**_F´ Version_**| current |
|**_Affected Component_**| Svc |
---
## Problem Description

ActiveTextLoggerComponent is `kind="active"`:
https://github.com/nasa/fprime/blob/devel/Svc/ActiveTextLogger/ActiveTextLoggerComponentAi.xml

However, it does not have an `async_input` port as required by the schematron spec:
https://github.com/nasa/fprime/blob/devel/Autocoders/Python/schema/default/active_comp_schematron.rng

## How to Reproduce

This was found by enabling exceptions for schematron validation failures (see #325).

## Expected Behavior

Change component so it passes schematron validation and passes gtests.

LeStarch, 2021-03-07:
This is a problem with the schematron.  This component has an "internal_interface".  The correct schema check is "must have async port or internal interface".

LeStarch, 2021-03-07:
See #455