nasa/fprime #451: command_res is active but does not have async_input port

billallen256, 2021-03-07:
| | |
|:---|:---|
|**_F´ Version_**| current |
|**_Affected Component_**| Autocoder |
---
## Problem Description

command_res is `kind="active"`:
https://github.com/nasa/fprime/blob/devel/Autocoders/Python/test/command_res/Test1ComponentAi.xml

However, it does not have an `async_input` port as required by the schematron spec:
https://github.com/nasa/fprime/blob/devel/Autocoders/Python/schema/default/active_comp_schematron.rng

## How to Reproduce

This was found by enabling exceptions for schematron validation failures (see #325).

## Expected Behavior

Change component so it passes validation and passes gtests.

LeStarch, 2021-03-07:
I think the correct fix is to make this line an "async_input" 

https://github.com/nasa/fprime/blob/eaf57af03f0c074881c3a83c0b5325cb9906fcae/Autocoders/Python/test/command_res/Test1ComponentAi.xml#L47

LeStarch, 2021-03-07:
See #455

closed 2021-03-09 by billallen256