nasa/fprime #452: interface1 is active but does not have async_input port

billallen256, 2021-03-07:
| | |
|:---|:---|
|**_F´ Version_**| current |
|**_Affected Component_**| Autocoder |
---
## Problem Description

interface1 component is of `kind="active"`:
https://github.com/nasa/fprime/blob/devel/Autocoders/Python/test/interface1/TestComponentAi.xml

However, it does not have an `async_input` port as required by the schematron spec.  In fact it has no ports at all.
https://github.com/nasa/fprime/blob/devel/Autocoders/Python/schema/default/active_comp_schematron.rng

## How to Reproduce

Found by enabling exceptions for schematron validation failures (see #325).

## Expected Behavior

Component should pass schematron validation an gtests.

LeStarch, 2021-03-07:
This is a problem with the schematron.  This component has an "internal_interface".  The correct schema check is "must have async port or internal interface".

LeStarch, 2021-03-07:
See #455

closed 2021-03-09 by billallen256