nasa/fprime #457: Math Component Tutorial: fprime-util build --ut Error

johnduffymsc, 2021-03-08:
Firstly, thank you for such a helpful tutorial.

I've reached the point of unit testing the MathSender component, 2.4.1.3.2, and when I run:

```fprime-util build --ut```

I receive the following error:

```make: *** No rule to make target 'Ref_MathSender_test_ut_ut_exe'. Stop.```

I'm running this command from ```Ref/MathSender/test/ut```. Any suggestions would be appreciated.

Best wishes

LeStarch, 2021-03-08:
The first step is to make sure you have a CMakeLists.txt file in MathSender and it has been included in Ref's CMakeList.txt.

Then make sure you are registering a unit test in your MathSender's CMakeLists.txt.  Here is a sample block for registering a UT, notice the Ai and C++ files have been added:

https://github.com/nasa/fprime/blob/d438a64a2fd30848b750f87069f8905395de1e48/Ref/SignalGen/CMakeLists.txt#L20-L25

johnduffymsc, 2021-03-08:
Thanks for your reply.

My CMakeLists.txt file in my MathSender folder is straight from the tutorial:
```
set(SOURCE_FILES
  "${CMAKE_CURRENT_LIST_DIR}/MathSenderComponentAi.xml"
  "${CMAKE_CURRENT_LIST_DIR}/MathSenderComponentImpl.cpp"
)
register_fprime_module()

set(UT_SOURCE_FILES
  "${CMAKE_CURRENT_LIST_DIR}/test/ut/main.cpp"
  "${CMAKE_CURRENT_LIST_DIR}/test/ut/Tester.cpp"
  "${CMAKE_CURRENT_LIST_DIR}/test/ut/TesterBase.cpp"
  "${CMAKE_CURRENT_LIST_DIR}/test/ut/GTestBase.cpp"
)
register_fprime_ut()
```
Is the double ```_ut``` in the ```Ref_MathSender_test_ut_ut_exe``` error message possible significant?

LeStarch, 2021-03-08:
@timcanham just pointed out that you are running it from MathSender/test/ut.  You should run it from the MathSender folder directly.

johnduffymsc, 2021-03-08:
Thanks, I think that will solve my problem.

But having run ```fprime-util generate --ut``` from the MathSender directory, I now have some other issues... reading the tutorial I've become confused as to what should be in ```main.cpp``` and ```Tester.hpp```, and that is throwing errors. I will re-read this section of the tutorial and work through it.

Many thanks

johnduffymsc, 2021-03-08:
Fixed, thank you.

The tutorial currently states:
```
You must first generate the unit test before you can build it. Enter:

fprime-util generate --ut

Once you generate the unit test, you can build the unit test by entering:

fprime-util build --ut

You can run the unit test by typing the following in the MathSender (not test/ut) directory:

fprime-util check
```
May I suggest a slight tweak to the documentation, along the lines of:
```
From the MathSender (not/ut) directory, you can generate, build and run the unit tests as follows:

You must first generate the unit test before you can build it. Enter:

fprime-util generate --ut

Once you generate the unit test, you can build the unit test by entering:

fprime-util build --ut

You can run the unit test by typing the following:

fprime-util check
```

closed 2021-03-08 by johnduffymsc