nasa/fprime #463: Porting Fprime to FreeRTOS

Ruben1729, 2021-03-15:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**|  |
---
## Feature Description

Our team is currently trying to port Fprime to FreeRTOS on an atmel board. We were unable to find a cmake toolchain to facilitate the process.

We were wondering if there is a way to gather all the source files generated by the Fprime autocoder in order to compile them through the compiler provided by Microchip. Which fprime-util command is responsible for running the autocoder and where could we find these resulting source files? 

## Rationale

Joshua-Anderson, 2021-03-15:
You'll need to create a custom cmake platform and toolchain file for the compiler. Then, you can build for that toolchain with `fprime-util build <toolchain>`

A couple examples that may help you get started:
- [cmake toolchain for arduino teensy](https://github.com/fprime-community/fprime-arduino/tree/master/cmake)
- [cmake toolchain for vxworks 6.7](https://github.com/fprime-community/fprime-vxworks)

Building a cmake toolchain for a new target can be a little tricky, but you should be able to follow generic cmake tutorials for how to setup a new platform/toolchain.

max19931, 2021-05-05:
there is a FreeRTOS branch which hasn't been updated since 2018!!

Joshua-Anderson, 2021-05-25:
Since this is more of a question than an F' specific issue, I'm going to close this, but feel free to reach with questions via the F' mailing list.

closed 2021-05-25 by Joshua-Anderson