nasa/fprime #464: BufferManager Race Condition

arizvi786, 2021-03-15:
| | |
|:---|:---|
|**_F´ Version_**| 1.5 |
|**_Affected Component_**|  Svc::BufferManager |
---
## Problem Description

Svc::BufferManager is a passive component with the following guarded input ports: bufferSendIn and bufferGetCallee. While using this component in a topology, such as F' Ref, the 2 guarded input ports can be connected to separate active components, such as groundIf and fileUplink: https://github.com/nasa/fprime/blob/devel/Ref/Top/RefTopologyAppAi.xml#L860-L867. This results in some member variables of Svc::BufferManager, such as padSize, allocatedSize and freeIndex, being written to from both contexts without any lock/guard protection when writing to these member variables. This can lead to a race condition if bufferSendIn and bufferGetCallee ports are invoked from 2 separate contexts simultaneously. 

## How to Reproduce

1. Theoretically there could be 2 active components where component A invokes bufferSendIn and component B invokes bufferGetCallee port respectively. If both components A and B continue invoking the ports simultaneously in a loop, then it should eventually result in the race condition realizing itself and corrupting a data value within the BufferManager component instance.

## Expected Behavior

FSW assertion on a corrupted value resulting from race condition described above

timcanham, 2021-03-15:
@arizvi786 The input ports are guarded, so they should be protected:

```xml
        <port name="bufferSendIn" data_type="Fw::BufferSend"  kind="guarded_input"    max_number="1">
        </port>

        <port name="bufferGetCallee" data_type="Fw::BufferGet"  kind="guarded_input"    max_number="1">
        </port>
```

arizvi786, 2021-03-15:
@timcanham they're guarded against repeated calls to the same port right? Wouldn't there be a race condition if the 2 ports are called simultaneously each from a separate task when both ports modify the same member variables?

timcanham, 2021-03-15:
@arizvi786 All guarded ports no matter the type share the same mutex, so different ones can be called and still protect the component state.

arizvi786, 2021-03-15:
@timcanham that's good, and yes I do see that in the ComponentAc.cpp file the same mutex is locked for all guarded ports. This is not an issue, I'll close the ticket. Thanks for clarification.

closed 2021-03-15 by arizvi786