nasa/fprime #471: Add an ENUM_LENGTH constant to FPrime enums

kbreed, 2021-03-29:
| | |
|:---|:---|
|**_F´ Version_**| |
|**_Affected Component_**| Autocoder |
---
## Automatically add an ENUM_LENGTH constant to F' enums.

Have the autocoder add a constant with the name ENUM_LENGTH to F' enums, similar to how SERIALIZABLE_SIZE works.

## Rationale

Needing to know the length of an enum list is a common need. Now is it up to developers to add a final member to the enum and expect that any maintaining developers would not accidentally add another member after that one.

timcanham, 2021-03-29:
The autocoder already puts a member `_MAX` to the end of generated enums that serves that purpose. With the C/C++ rules for assigning values to enum members, it should always be the highest value. If none of the members have explicit values assigned, it can be used for the size as well.

saba-ja, 2021-03-29:
I believe the above request is referring to [enums/enum_hpp.tmpl](https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/generators/templates/enums/enum_hpp.tmpl ) rather than [component/hpp.tmpl](https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/generators/templates/component/hpp.tmpl)

## Clarification - Part 1:

`${enum_type}_MAX` is implemented in [component/hpp.tmpl](https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/generators/templates/component/hpp.tmpl) which adds `_MAX` to the enums defined in `*ComponentAi.xml`

**Example:**

Assuming one has a component named `ActivityManagerComponentAi.xml` with the following command definition:
``` xml
<commands>
        <command kind="guarded" opcode="0" mnemonic="Recording">
            <comment>
            Command to start recording science data to file
            </comment>
            <args>
            <arg type="ENUM" name="ARG1">
                <enum name="rec_start_top">
                    <item name="start"/>
                    <item name="stop"/>
                </enum>
            </arg>
            </args>
        </command>
...
</commands>
```

The autocoder will generate the following `*Ac.hpp` file:
`ActivityManagerComponentAc.hpp`

```c++
...
    typedef enum {
      start, /*  */
      stop, /*  */
      rec_start_top_MAX
    } rec_start_top;

```

In this case, as mentioned above, `rec_start_top_MAX` value can be used as the number of items in the enum.

## Clarification - Part 2:

The above request is referring to a case where user has defined an `*EnumAi.xml` file and wanted to know
the number of enum items.

**Example:**

Assume one has the following enum definition:

`InstanceNameEnumAi.xml`

```xml
<?xml version="1.0" encoding="UTF-8"?>
<?oxygen RNGSchema="file:../../Autocoders/Python/schema/ISF_Type_Schema.rnc" type="compact"?>
<enum namespace = "Backplane" name="InstanceName">
    <comment>Enum of all the components on the backplane</comment>
    <item name="INVALID" value="-1"/>
    <item name="BP_MANAGER" value="0"/>
    <item name="C4D" value="1"/>
    <item name="CSM" value="2"/>
    <item name="CE_C4D" value="3"/>
    <item name="CE_LIF" value="4"/>
    <item name="CE_ESI_MS" value="5"/>
    <item name="FPM" value="6"/>
    <item name="ELVIS_ILL" value="7"/>
    <item name="ELVIS_SPU" value="8"/>
    <item name="EXTRACTOR" value="9"/>
    <item name="LIF" value="10"/>
    <item name="THERMAL" value="11"/>

    <!-- Keep this item last and update value when updated-->
    <item name="NUM_INSTANCES" value="12"/>
</enum>
```

As shown above the user is manually adding a last member to keep track of the number of enum items.
This is a code smell since the developer has to manually keep track of the number items.

Since autocoder enum template [enums/enum_hpp.tmpl](https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/generators/templates/enums/enum_hpp.tmpl ) (shown below):
https://github.com/nasa/fprime/blob/9ba5e0e8600823f3d8b9078b89bd3aff43e0beea/Autocoders/Python/src/fprime_ac/generators/templates/enums/enum_hpp.tmpl#L68-L70

is already aware of the number of members it could be beneficial to add the following to the [enums/enum_hpp.tmpl](https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/generators/templates/enums/enum_hpp.tmpl ) template to allow user keep track of the number of enum items.
```
     enum {
         SERIALIZED_SIZE = sizeof(FwEnumStoreType),
         ENUM_LENGTH = ${len($items_list)} <--------------- add this to keep track of number of enum items
         };
```  

In the case above, the resulted autogenerated `EnumAc.hpp` will be the following:

`InstanceNameEnumAc.hpp`
```c++
  ...
    enum {
      SERIALIZED_SIZE = sizeof(FwEnumStoreType),
      ENUM_LENGTH = 14
    };
```
where user can get number of items by `InstanceName::ENUM_LENGTH`.

Other approach could be to follow the same pattern as  [component/hpp.tmpl](https://github.com/nasa/fprime/blob/devel/Autocoders/Python/src/fprime_ac/generators/templates/component/hpp.tmpl)
 and add a `_MAX` item as the last item in the following:
https://github.com/nasa/fprime/blob/9ba5e0e8600823f3d8b9078b89bd3aff43e0beea/Autocoders/Python/src/fprime_ac/generators/templates/enums/enum_hpp.tmpl#L44-L60

timcanham, 2021-03-29:
It seems we should have the external enum autocoder match the output of the component enum.

dav-oss, 2021-04-03:
That seems to be a good idea

saba-ja, 2021-05-14:
Added by #477

closed 2021-05-14 by saba-ja